30350n/pcb2blender #1: The exporter does not work correctly

HaydenHu, 2022-06-25:
[image]

30350n, 2022-06-25:
Should be fixed in the [v1.2 release](https://github.com/30350n/pcb2blender/releases/tag/v1.2-k6.0-b3.1). Can you try that one and confirm that it works now?

HaydenHu, 2022-06-25:
> Should be fixed in the [v1.2 release](https://github.com/30350n/pcb2blender/releases/tag/v1.2-k6.0-b3.1). Can you try that one and confirm that it works now?

The exporter is ready to use, but the run icon does not load correctly. And after the importer is installed in blender3.1/3.2, it will freeze when I enable it, causing blender to crash

30350n, 2022-06-26:
Made another release ([v1.3](https://github.com/30350n/pcb2blender/releases/tag/v1.3-k6.0-b3.2)) to fix that.
Blender freezing when enabling the addon for the first time is normal, because its installing some dependencies.
It shouldn't throw an error/crash though.

HaydenHu, 2022-06-26:
> 制作了另一个版本（[v1.3](https://github.com/30350n/pcb2blender/releases/tag/v1.3-k6.0-b3.2)）来解决这个问题。 首次启用插件时 Blender 冻结是正常的，因为它安装了一些依赖项。 它不应该抛出错误/崩溃。

[image]
[image]

30350n, 2022-06-26:
urgh .... seems like cairosvg requires a bunch of dlls that don't get installed via pip.
For now you should be able to fix that by installing a GTK+ runtime https://github.com/tschoonj/GTK-for-Windows-Runtime-Environment-Installer (mingw also comes with it I think). That's not a good/permanent solution though, will have to use something else to rasterize the gerbers/svgs I guess ...

HaydenHu, 2022-06-27:
> urgh .... seems like cairosvg requires a bunch of dlls that don't get installed via pip. For now you should be able to fix that by installing a GTK+ runtime [...]

Yes, but the importer option only appears once when the plugin is enabled, disappears if it is closed and turned back on, and needs to be enabled again for the plugin to appear

30350n, 2022-06-27:
> Yes, but the importer option only appears once when the plugin is enabled, disappears if it is closed and turned back on, and needs to be enabled again for the plugin to appear
That sounds odd. If the plugin registers fine once, I don't really see why it wouldn't register again fine later aswell...
Are there any errors in the system console (`Window -> Toggle System Console` in Blender) when that happens?

HaydenHu, 2022-06-27:
> That sounds odd. If the plugin registers fine once, I don't really see why it wouldn't register again fine later aswell... Are there any errors in the system console (`Window -> Toggle System [...]

[image]

30350n, 2022-06-27:
Thanks, fixed on master for now. Will make another release as soon as I figured out a better replacement for cairo.

30350n, 2022-06-27:
I completely got rid of cairo now, so all the GTK+ installation stuff shouldn't be required anymore. I also fixed the issue with the icon. Can you test the [1.4 release](https://github.com/30350n/pcb2blender/releases/tag/v1.4-k6.0-b3.2) and see if everything works now?

HaydenHu, 2022-06-28:
> 我现在完全摆脱了 cairo，所以不再需要所有 GTK+ 安装的东西了。我还解决了图标的问题。你能测试一下[1.4 版本](https://github.com/30350n/pcb2blender/releases/tag/v1.4-k6.0-b3.2)，看看现在是否一切正常吗？

The pcb3d imported by blender has only PCB and no electronic component model
[image]
[image]

30350n, 2022-06-28:
Any errors in the system console after importing?
Also, are the components present in the KiCad 3D viewer?

HaydenHu, 2022-06-28:
> 导入后系统控制台中是否有任何错误？ 此外，KiCad 3D 查看器中是否存在组件？

[image]
[image]

30350n, 2022-06-28:
Try reexporting the file with the new v1.4 exporter.

30350n, 2022-06-28:
Any more errors?
If not, can you share the `.pcb3d` file?

HaydenHu, 2022-06-28:
[123.zip](https://github.com/30350n/pcb2blender/files/8998916/123.zip)
I didn't find any other bugs, only the version number didn't change and it's still 1.1.0

30350n, 2022-06-28:
And this one is also fixed in [v1.5](https://github.com/30350n/pcb2blender/releases/tag/v1.5-k6.0-b3.2).
Thanks a lot for reporting all the issues and all your testing! ❤️

closed 2022-06-28 by 30350n