30350n/pcb2blender #4: can not enable blender addon

chenxuuu, 2022-06-27:
I installed `gtk3-runtime-3.24.31-2022-01-04-ts-win64` , but it show this

[image]

[image]

30350n, 2022-06-27:
Make sure that that directory is in your PATH variable.

chenxuuu, 2022-06-27:
it **IS** in PATH variable **and** blender addon **SCANED THIS PATH** in screenshot

chenxuuu, 2022-06-27:
[image]

30350n, 2022-06-27:
hmmm, yeah not sure what that's about tbh ...
Seems to be some kind of issue with cairo, not going to spend any time on investigating that further though, as I'm currently working on completely removing cairo as a dependency all together.