30350n/pcb2blender #5: macOs support?

2kohm, 2022-07-29:
Hello, i followed the install instructions and the plugin is visible in plugin manager.
[image]
but i don't have any export button in kicad.

30350n, 2022-07-29:
Uhm ... what instructions did you follow/how did you install it exactly?

30350n, 2022-08-07:
@2kohm did you manage to resolve this?

2kohm, 2022-08-07:
hi, sadly not.

(manual) Download the pcb2blender_exporter.zip from the [latest release](https://github.com/30350n/pcb2blender/releases/latest), unpack the plugins directory from it into your [Kicad Plugin directory](https://dev-docs.kicad.org/en/python/pcbnew/) and rename it to pcb2blender_exporter.

30350n, 2022-08-07:
Do you get any errors/popups when you open the PCB Editor after installing the plugin?
If not, can you also check what python version your KiCad is using, as shown below?
[image]

Frzoen, 2022-08-07:
Hi again. Related to #6,
Just so happened that I've, for a brief moment, had Mac in my hands, so what I did? Of course checked version of Python :)
So, on macOS 12.04 default installation of KiCad is bundled with `Python 3.8.2`.
Importing plugin using content manager is not working. Installing it manually according to Your instruction allows for `import pcb2blender_exporter` in KiCad console.
Calling it however yields same error as on Ubuntu 20.04:
```
>>> import pcb2blender_exporter
Traceback (most recent call last):
  File "<input>", line 1, in <module>
ModuleNotFoundError: No module named 'pcb2blender_exporter'
import pcb2blender_exporter
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "/Users/user_name/Documents/KiCad/6.0/scripting/plugins/pcb2blender_exporter/__init__.py", line 5, in <module>
    from .export import export_pcb3d, get_boarddefs
  File "/Users/user_name/Documents/KiCad/6.0/scripting/plugins/pcb2blender_exporter/export.py", line 24, in <module>
    class StackedBoard:
  File "/Users/user_name/Documents/KiCad/6.0/scripting/plugins/pcb2blender_exporter/export.py", line 26, in StackedBoard
    offset: list[float]
TypeError: 'type' object is not subscriptable
```
Using `from typing import List` fixed issue.
Didn't checked flow with blender tho.

30350n, 2022-08-07:
Ah fuck it. I've made it use `List` again in the [v1.7 release](https://github.com/30350n/pcb2blender/releases/tag/v1.7-k6.0-b3.2) now again. Hopefully that just makes everything work on Linux/macOS without requiring any messing with python versions or self patching the addon.
I don't really like doing it this way, but it seems like the best compromise for now.

30350n, 2022-08-07:
@2kohm can you try using [v1.7](https://github.com/30350n/pcb2blender/releases/tag/v1.7-k6.0-b3.2) and report back if that fixes the issue?

30350n, 2022-10-04:
Closing this because of inactivity and because from what I can tell the issue might've already been fixed.

closed 2022-10-04 by 30350n