30350n/pcb2blender #6: Plugin not showing up in PCB Editor

Frzoen, 2022-08-07:
As in title. 
Running Ubuntu 20.04
Installed both manually and via content manager ( Plugin And Content Manager -> Install from File...)
After that extension didn't showed up in PCB editor.

I've traced issue to this:
```
>>> import pcb2blender_exporter
Traceback (most recent call last):
  File "/usr/lib/python3.8/code.py", line 90, in runcode
    exec(code, self.locals)
  File "<input>", line 1, in <module>
  File "/home/user_name/.local/share/kicad/6.0/scripting/plugins/pcb2blender_exporter/__init__.py", line 5, in <module>
    from .export import export_pcb3d, get_boarddefs
  File "/home/user_name/.local/share/kicad/6.0/scripting/plugins/pcb2blender_exporter/export.py", line 24, in <module>
    class StackedBoard:
  File "/home/user_name/.local/share/kicad/6.0/scripting/plugins/pcb2blender_exporter/export.py", line 26, in StackedBoard
    offset: list[float]
TypeError: 'type' object is not subscriptable
```

To fix it I've changed 
``` 
import pcbnew
from pcbnew import PLOT_CONTROLLER as PlotController, PCB_PLOT_PARAMS, PLOT_FORMAT_SVG, ToMM

import tempfile, shutil, struct, re
from pathlib import Path
from zipfile import ZipFile, ZIP_DEFLATED
from dataclasses import dataclass, field

PCB = "pcb.wrl"
COMPONENTS = "components"
LAYERS = "layers"
LAYERS_BOUNDS = "bounds"
BOARDS = "boards"
BOUNDS = "bounds"
STACKED = "stacked_"

INCLUDED_LAYERS = (
    "F_Cu", "B_Cu", "F_Paste", "B_Paste", "F_SilkS", "B_SilkS", "F_Mask", "B_Mask"
)

SVG_MARGIN = 1.0 # mm

@dataclass
class StackedBoard:
    name: str
    offset: list[float]

@dataclass
class BoardDef:
    name: str
    bounds: list[float]
    stacked_boards: list[StackedBoard] = field(default_factory=list)
```

to 
```
import pcbnew
from pcbnew import PLOT_CONTROLLER as PlotController, PCB_PLOT_PARAMS, PLOT_FORMAT_SVG, ToMM

import tempfile, shutil, struct, re
from pathlib import Path
from zipfile import ZipFile, ZIP_DEFLATED
from dataclasses import dataclass, field
from typing import List

PCB = "pcb.wrl"
COMPONENTS = "components"
LAYERS = "layers"
LAYERS_BOUNDS = "bounds"
BOARDS = "boards"
BOUNDS = "bounds"
STACKED = "stacked_"

INCLUDED_LAYERS = (
    "F_Cu", "B_Cu", "F_Paste", "B_Paste", "F_SilkS", "B_SilkS", "F_Mask", "B_Mask"
)

SVG_MARGIN = 1.0 # mm

@dataclass
class StackedBoard:
    name: str
    offset: List[ float ]

@dataclass
class BoardDef:
    name: str
    bounds: List[float]
    stacked_boards: List[StackedBoard] = field(default_factory=list)
```
Later I've found that there are Python3.10 dependencies in blender import script. Maybe going to Python3.10 would also fix extension not showing up in PCB Editor. I can check later.

30350n, 2022-08-07:
Just added a note to the README about this. Basically KiCad and Blender both come with their own bundled python version on Windows, so whatever version they come with, is generally the one I'll support. (For KiCad 6.0 it's currently python 3.9, for Blender 3.2 it's python 3.10).
Can you update your systems python version to 3.10 and see if any issues persist?

Frzoen, 2022-08-07:
For me KiCad6 is loading Python 3.8.10 by default and tbh I don't know how to change it without re-compiling KiCad. Possibly, as You've stated in a note, this is distro related.
```
Py 0.9.8
Python 3.8.10 (default, Jun 22 2022, 20:18:18) 
[GCC 9.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
Startup script executed: /home/user_name/.config/kicad/6.0/PyShell_pcbnew_startup.py
>>> 
```

I've also checked that I've set up Python 3.10 as system default both for `python` and `python3` commands
```
$ user_name@host_name:~$ python --version
Python 3.10.6
$ user_name@host_name:~$ python3 --version
Python 3.10.6
```

Anyway, adding this change would be quick and script will work with Python 3.8.10 (as it seems to be default combo with KiCad6 on Ubuntu <- I might be wrong on that).
Additionally I've checked flow with KiCad->Blender export. Looks like great tool, works like a charm :) Nice work!

Frzoen, 2022-08-07:
Checked Flatpak version of KiCad, it comes with `Python 3.9.9 (main, Nov 10 2011, 15:00:00)` and export works out of the box.

30350n, 2022-08-07:
Interesting, well I guess that works then ¯\_(ツ)_/¯.
I also just checked the package registry for 20.04 and it says it still ships KiCad 5.1, which might be even more problematic possibly, but yeah. Sounds like you managed to get it to work fine, so ... have fun!

closed 2022-08-07 by 30350n