30350n/pcb2blender #11: Plugin crash in KiCad 7.0.0

oliv33bx, 2023-02-16:
once installed well, the export plugin doesn't work with the latest version of KiCad 7.0.0

set-soft, 2023-02-17:
KiBot can export *pcb3d* files using code based on this project. It can currently do it for KiCad 7 (*dev* branch, not yet released because KiCad 7.0.0 has some important bugs in the Python API). While adapting the code I remember I had to:
- Replace `pad.GetOrientationRadians()` by `pad.GetOrientation().AsRadians()`
- Be careful with the SVG view boxes. They must support floating point values because now KiCad uses millimeters as SVG units. If you use integers the layers doesn't get centered and you'll see the silkscreen and copper with some offset in the rendered board.

30350n, 2023-02-18:
Fixed in [v2.2](https://github.com/30350n/pcb2blender/releases/tag/v2.2-k7.0-b3.4), thanks a lot!
Thanks for the hints @set-soft. Getting the svg bounding boxes right again took some more work actually, not sure how you implemented it in KiBot, but [here](https://github.com/30350n/pcb2blender/blob/b338ac0bce7ec3b03639f6adff0ef64d0a05b8ca/pcb2blender_exporter/export.py#L45) and [here](https://github.com/30350n/pcb2blender/blob/b338ac0bce7ec3b03639f6adff0ef64d0a05b8ca/pcb2blender_exporter/export.py#L192) is the relevant code if you want to check it out.

closed 2023-02-18 by 30350n

oliv33bx, 2023-02-18:
Thanks @30350n  for the new version of exporter-importer.
However, I tried to install the importer plugin in blender 3.4 but it does not appear in the plugin list.

set-soft, 2023-02-20:
> Fixed in [v2.2](https://github.com/30350n/pcb2blender/releases/tag/v2.2-k7.0-b3.4), thanks a lot! Thanks for the hints @set-soft. Getting the svg bounding boxes right again took some more work [...]

Your code looks ok to me. KiBot uses its own code to generate the SVG, not the code from PCB2Blender.

The SVG is just another output format. So the code is different. It computes the bounding box using KiCad IUs (internal Units) and then converts it to SVG units.

The conversion support KiCad 5.1, 6.0 and 7.0. All of them uses different SVG units, in 6.0 they are configurable.

The relevant code is:

```python
        # Limit the view box of the SVG
        bbox = GS.get_rect_for(GS.board.ComputeBoundingBox(self.size_detection == 'kicad_edge'))
        # Apply the margin (left right top bottom)
        bbox = (bbox[0]-self.margin[0], bbox[1]-self.margin[2],
                bbox[2]+self.margin[0]+self.margin[1], bbox[3]+self.margin[2]+self.margin[3])
        # Width/height of the used area in cm
        width = ToMM(bbox[2])*0.1
        height = ToMM(bbox[3])*0.1
        # Scale factor to convert KiCad IU to the SVG units
        bbox = GS.iu_to_svg(bbox, self.svg_precision)
        logger.debug('Adjusting SVG viewBox to {} for width {} cm and height {} cm'.format(bbox, width, height))
        for f in self._generated_files.values():
            fname = os.path.join(output_dir, f)
            logger.debugl(2, '- '+f)
            change_svg_viewbox(fname, bbox, width, height)
```

And the SVG patcher:

```python
SVG_VIEW_BOX_REGEX2 = r'width="(.*)" height="(.*)" viewBox="(\S+) (\S+) (\S+) (\S+)"'
SVG_VIEW_BOX_SUB_PAT = r'width="{}cm" height="{}cm" viewBox="{} {} {} {}"'

def change_svg_viewbox(file, view_box, w, h):
    with open(file, 'rt') as f:
        text = f.read()
    text = re.sub(SVG_VIEW_BOX_REGEX2, SVG_VIEW_BOX_SUB_PAT.format(w, h, view_box[0], view_box[1], view_box[2], view_box[3]),
                  text)
    with open(file, 'wt') as f:
        f.write(text)
```

I keep the width and height in cm to generate files as similar as possible to the originals.