30350n/pcb2blender #12: Importer installation issue in Blender 3.4

oliv33bx, 2023-02-18:
Thanks @30350n  for the new version of exporter-importer.
However, I tried to install the importer plugin in blender 3.4 but it does not appear in the plugin list.

_Originally posted by @oliv33bx in https://github.com/30350n/pcb2blender/issues/11#issuecomment-1435660667_

30350n, 2023-02-18:
What OS are you on?

oliv33bx, 2023-02-18:
Windows 10

30350n, 2023-02-18:
How are you installing the plugin exactly? (just describe what you're doing, step by step)

oliv33bx, 2023-02-18:
In Blender, I select edit/preferences/add-on/install
I choose the doswnloaded file pcb2blender_importer_2-2.zip and click install.
The plugin appears in the non active plugins list and when I try to activate it, an error occurs
I tried to remove the plugin in order to re-intall it but an error occurs and I can not remove the plugin (PermissionError...)

oliv33bx, 2023-02-18:
Finally I managed to uninstall and reinstall the importer plugin and now it works !
Sorry for the inconvenience !

closed 2023-02-18 by oliv33bx