30350n/pcb2blender #15: I need to disable/enable the importer addon in blender every time I start blender

Kedarius, 2023-04-07:
I had a really strange issue where the board and some ICs had missing materials. After some finagling I've found out that the issue is that the importer plugin seems to be enabled (enabled checkbox is checked) but there is no Import from PCB and most importantly there is no PCB Shader.

All I need to do is disable/enable the addon and everything works until I restart blender. Then I need to disable/enable it again.

U have it installed from zip file as described and I reinstalled Blender to current version.

[system-info.txt](https://github.com/30350n/pcb2blender/files/11181059/system-info.txt)

It is not a big deal once I've figured it out but as I am a blender newbie it took me quite some time to figure it out....

30350n, 2023-04-07:
Sounds like you might be using Blender 3.5? The addon currently doesn't support 3.5 because of a bug in Blender.

Kedarius, 2023-04-07:
Oh yeah, I just downloaded it so I have the current version and it is 3.5. 
I have to say that apart from the mentioned disable/enable bug it seems to be working good though :-)

closed 2023-04-07 by Kedarius

30350n, 2023-04-07:
Thanks! If you run into any other issues, feel free to report them ^^

You can also try a 3.6 Alpha build or a 3.5.1 RC build from https://builder.blender.org/download/daily/archive/ btw, the alpha build definitely has the fix, not sure if it got backported to 3.5 already.