30350n/pcb2blender #16: PCB turns black after saving

mtribiere, 2023-04-25:
# Environment 

PCB2Blender v2.4
Blender 3.4
Kicad 7.0.1

# Step to reproduce

- Export a PCB from Kicad
- Load it into Blender
- Save the scene to a .blend file
- Close Blender
- Reopen the scene

# What happens
The PCB in the scene turn completely black
[image]

# What should happen
The PCB should appear normally

mtribiere, 2023-04-25:
Probably a duplicate of #15, since the workaround worked too. 

Summary:
- Before opening the scene, disable/enable the add-on
- Then open the scene
- Voilà the textures are back

30350n, 2023-04-25:
Hm this is odd, this shouldn't be happening in 3.4.
Before disabling/enableing the addon, are the any errors in the system console?
If not, can you share a screenshot of the node tree?

30350n, 2023-04-25:
Also, [here](https://github.com/30350n/pcb2blender/files/11326276/pcb2blender_importer_2-6.zip) is a prebuild for Blender 3.5.1 (there is a bug in 3.5.0 which made the addon hard crash blender).
I'd be curious if you get the same behavior there.

mtribiere, 2023-04-26:
There is indeed an error in the system console before I load my scene:
```
Read prefs: C:\Users\Matthieu\AppData\Roaming\Blender Foundation\Blender\3.4\config\userpref.blend
Traceback (most recent call last):
  File "C:\Program Files\Blender Foundation\Blender 3.4\3.4\scripts\modules\addon_utils.py", line 333, in enable
    mod = __import__(module_name)
  File "C:\Users\Matthieu\AppData\Roaming\Blender Foundation\Blender\3.4\scripts\addons\pcb2blender_importer\__init__.py", line 22, in <module>
    add_dependencies(deps, no_extra_deps=True)
  File "C:\Users\Matthieu\AppData\Roaming\Blender Foundation\Blender\3.4\scripts\addons\pcb2blender_importer\blender_addon_utils\register.py", line 78, in add_dependencies
    if not importlib.util.find_spec(module):
AttributeError: module 'importlib' has no attribute 'util'

```

mtribiere, 2023-04-26:
Just installed and tried on Blender 3.5.1, I can confirm the pre-release fixed the issue

30350n, 2023-04-26:
Ah yeah the weird importlib bug, I fixed that in the 3.5 version. Maybe I should backport that to 3.4 ...

mtribiere, 2023-04-26:
Maybe open an issue in Blender? Must be crowded place though

mtribiere, 2023-04-26:
Make sense, I leave the issue open then so you can close it with your commit

30350n, 2023-05-17:
This is fixed in the [v2.6](https://github.com/30350n/pcb2blender/releases/tag/v2.6-k7.0-b3.5) release.

closed 2023-05-17 by 30350n