30350n/pcb2blender #17: AttributeError: 'NoneType' object has no attribute 'data'

frippe75, 2023-04-30:
Not sure if this is a symptom of VRML export problem or with the pcb2blender??

```
ValueError: invalid mode: 'rU'
Failed to open file: /tmp/pcb2blender_tmp/PCB_LEDBEE_pcb3d_6426931431228453188/components/ESP32-S3-MINI-1.wrl
Traceback (most recent call last):
  File "/home/frta/.config/blender/3.4/scripts/addons/pcb2blender_importer/importer.py", line 156, in execute
    if (pcb := self.import_pcb3d(context, filepath)) == {"CANCELLED"}:
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/frta/.config/blender/3.4/scripts/addons/pcb2blender_importer/importer.py", line 299, in import_pcb3d
    obj.data.name = component.rsplit("/", 1)[1].rsplit(".", 1)[0]
    ^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'data'

```

frippe75, 2023-04-30:
Renamed the pcb2blender file to .zip. unzip and moved to the components dir. 
Uploaded the ESP32-S3-mini to an online viewer.

[image]

frippe75, 2023-04-30:
Trying to look at the bounds file but listed as data... 

```
$ hexdump bounds 
0000000 0443 8bbd a842 f5db 1442 0ebe 1442 098a
0000010

```
But not sure how the PCB outline is realized as an object. Maybe I set the text elements too tight? Not sure how that works.

frippe75, 2023-04-30:
$ python --version
Python 3.11.3

30350n, 2023-04-30:
I doubt the file contents are the actual problem. Probably just some edge case I missed. Can you share the .pcb3d file? Also knowing the importer version and blender version would be helpful :)

frippe75, 2023-04-30:
Blender 3.4.1, importer is from latest release pcb2blender_importer_2-4.zip.
Blender 3.5 seems to have a packaging problem where no plugins are loaded so don't go there.

[PCB_LEDBEE2.pcb3d.gz](https://github.com/30350n/pcb2blender/files/11361926/PCB_LEDBEE2.pcb3d.gz)

frippe75, 2023-04-30:
Tried installing the debugger for VS code to have a look at the issue myself but got stuck in actually getting that working. Debugger never starts listening on that port.

30350n, 2023-04-30:
> Not sure if this is a symptom of VRML export problem or with the pcb2blender??
>
> ```
> ValueError: invalid mode: 'rU'
> Failed to open file: [...]

Is this the full traceback?

I'm not sure what `ValueError: invalid mode: 'rU'` is about really, but `Failed to open file: .../ESP32-S3-MINI-1.wrl` definitely seems problematic. Not sure why that happens though tbh.

Just tested the file and on my machine it just works fine.
[image]

Sooooo, this is probably some OS problem I'd say. Are you on linux or mac?

frippe75, 2023-05-01:
Ohh. On Fedora38, Blender 3.4.1. What are you on since it appear to work??

frippe75, 2023-05-01:
To me, it appears that the passed context is not valid for some reason.

[context.object is None in execute function](https://blenderartists.org/t/context-object-is-none-in-execute-function/634552)

Not sure why yet

frippe75, 2023-05-01:
Inserted a breakpoint and looked at the context passed to importerp.py
```

>>> print(context)
<bpy_struct, Context at 0x7f1543f0b168>
>>> print(context.object)
<bpy_struct, Object("Cube") at 0x7f153c0d2008>
>>> print(context.object.data)
<bpy_struct, Mesh("Cube") at 0x7f153bcc7608>
```

The context is the default cube when starting blender. 

At what stage or in which context do you run your import? I think the addon might have to set it if it matters.

frippe75, 2023-05-01:
@30350n would it be possible to ask you for the blend file?

30350n, 2023-05-01:
> Ohh. On Fedora38, Blender 3.4.1. What are you on since it appear to work?? Especially your Python version would be intereting.

Windows 10 and I'm testing on Blender 3.5 atm, which ships with Python 3.10.9.

> To me, it appears that the passed context is not valid for some reason.

Yeah, but that's not the root of the problem. The context is invalid because the `.wrl` importer fails with
```
ValueError: invalid mode: 'rU'
Failed to open file: /tmp/pcb2blender_tmp/PCB_LEDBEE_pcb3d_6426931431228453188/components/ESP32-S3-MINI-1.wrl
```
resulting in an invalid context later on I think.

> @30350n would it be possible to ask you for the blend file?

Sure, but it won't really be helpful for debugging. (This is exported from 3.5, but you'll need 3.5.1 to be able to open it.)
[LEDBEE.zip](https://github.com/30350n/pcb2blender/files/11365137/LEDBEE.zip)

30350n, 2023-05-17:
Did you have any chance to look into this? Would be interesting what's the cause exactly. Maybe it actually is an issue induced by Python 3.11.

30350n, 2023-05-19:
Just managed to run into this on linux aswell, seems to be a bug with the blender .wrl importer that has already been fixed but not backported to 3.5. Will be fixed in 3.6 probably.

closed 2023-05-19 by 30350n