30350n/pcb2blender #19: Floating solder blobs...

UnexpectedMaker, 2023-05-17:
I am seeing many of the little solder blobs you generate floating above the pads, either half way up a component, or above them, like with the 0201 resistors you can see in this image....
[image]

The 0201 resistors are just the standard models in KiCAD, as are the 0402 ones that don't have this issue, but also don't have solder blobs.

I'm unsure how they are detected to be generated, or why they might be missing or floating.

Any idea what I need to do in my model or data to solve this?

Thanks :)

30350n, 2023-05-18:
Hmm this is odd, could it be that this is a <1.6mm board? I think I might have 1.6mm hard coded as the pcb thickness somewhere 😅 . That wouldn't explain why it seemingly only happens for some components though? Odd.
Would you mind sharing the .pcb3d file so I could take a look?

UnexpectedMaker, 2023-05-19:
Yeah, it's a 1mm thickness board. Is there a way I can share the board with you, but not publicly?

30350n, 2023-05-19:
Sure, you can send it to mail@30350n.de if you want .... but actually I had some time on hand so I implemented basic stackup support ^^.

Here's a prerelease for the exporter and importer (you need to reexport the .pcb3d files) which should fix the pcb thickness issues, as well as automatically use your defined stackup colors:
[pcb2blender_exporter_3-0.zip](https://github.com/30350n/pcb2blender/files/11518011/pcb2blender_exporter_3-0.zip)
[pcb2blender_importer_3-0.zip](https://github.com/30350n/pcb2blender/files/11518012/pcb2blender_importer_3-0.zip)

UnexpectedMaker, 2023-05-19:
Wicked, trying this now!

UnexpectedMaker, 2023-05-19:
Ok, that seems to fix it and it brought in the correct board thickness and colour!
Thanks heaps :)
[image]

closed 2023-05-20 by UnexpectedMaker