30350n/pcb2blender #20: Auto PCB shape detection?

UnexpectedMaker, 2023-05-20:
Me again :)

Curious as to why we always need to pin the TL and BR areas of the PCB for detection of the area?

I get that allows you to scrape the name we want to use for the board, and very useful or extracting just an area of a design, but can it not just default to the EdgeCuts outline are if no TL and BR areas are marked?

I think it would be a big time saver - especially considering I am wanting to export my entire product list of boards :)

30350n, 2023-05-21:
Oh, you don't need to do that at all actually, it already works as you described ;)

The board marking feature is mainly intended multi board designs. Say you've got one schematic, split up over two boards (for space reasons), connected with standard pin headers/sockets. The feature allows you to define the outline of those two boards, as well as their relation to each other, so that when you import the .pcb3d file in Blender, they automatically get cut (to remove potential mouse bites etc.) and stacked on each other.

I should probably work on the wording of the KiCad tooltip a little more ... or maybe just finally make a proper guide on how to use all of this ^^

UnexpectedMaker, 2023-05-21:
Really? wow, ok, I thought I tried it a few versions ago and it didn't export anything, so I added the tags and then it did.
Maybe I did something else stupid at the time ;)
Thanks for confirming.

closed 2023-05-21 by UnexpectedMaker