30350n/pcb2blender #21: Is there a way to include the inner layers into the .pcb3d model?

PurpleLobster, 2023-05-25:
I would like to make a render of the internal layers of the PCB (fully transparent board material).
Is there a way to include the inner layers into the .pcb3d model?

30350n, 2023-05-26:
Not with the rasterized PCB material at least. If KiCAD exports inner layers in their .wrl files the "3D (deprecated)" material mode might work, but you'll have to spend some time adjusting the materials.

PurpleLobster, 2023-06-06:
I checked it out, unfortunately KiCad does not export the inner layers in their .wrl files :(
Does pcb2blender use KiCad's .wrl export function is it based on .svg export?

30350n, 2023-06-07:
Both. The .wrl is being used as a base, for the main PCB body and components and .svg files are being used for the shading. Including inner layers is definitely something that can be done, but it would require some work to be implemented.