30350n/pcb2blender #22: Inconsistency with solder blob generation for pads....

UnexpectedMaker, 2023-05-28:
Some components have the "generated solder blobs" for the pads and some components don't - possibly only the KiCAD built in model do?

I've highlighted some good areas in green and bad in red as an example.
[image]

 How do I get them to be added to all components? Is there something teh script is looking for in the component name? or in a property setting somewhere?

Thanks :)

30350n, 2023-05-28:
The conditions for "SMART" solder joint generations are as follows:
1. The component has to have a 3d model and be either THT or SMD
2. The pad has to
    - have the according Paste layer selected as a technical layer
    - be one of these pad shapes: CIRCLE, OVAL, RECT, ROUNDRECT
  
You can check if the Paste layer is enabled in KiCad here:
[image]

If it's an unsupported pad shape, there should be a warning in the system console in Blender.

UnexpectedMaker, 2023-05-28:
Ok, perfect, thanks!
Seems I had a bunch of footprints set as "Unspecified" - whoops!
Love your work :)

closed 2023-05-28 by UnexpectedMaker