30350n/pcb2blender #24: More optimised or simpler shaders?

UnexpectedMaker, 2023-06-27:
I just got myself a new Mac with an M2 Max chip, and though the shaders are much faster, they are still prohibitively slow and I remember you mentioned you were working on (or thinking of working on) more simplified shaders?

Is that something that is happening and is it near?

Thanks
Seon

30350n, 2023-06-27:
I've spent some time on this, but haven't gotten far enough to release anything yet. Currently working on other projects, so not much progress, I hope I can finish this in the next few months.
I'm not sure about how well apple silicon is supported in Blender tbh, but for proper Cycles rendering a dedicated (nvidia) GPU is probably the best option.
Although I remember having atleast somewhat useable results on my Thinkpad too (2nd gen ryzen).

UnexpectedMaker, 2023-06-27:
Ok, nps. No option for a dedicated GPU on a Mac anymore, ad buying a PC just to get a discreet GPU just for this is also not an option ;)

Blender is native on Apple Silicon now, but more work is being done between Blender and Apple to fix bugs and optimise more, but I feel that for what I am seeing is my render views, most could look "close to the same" with simpler shaders, but this *really* isn't my area of enterprise ;)

hstarmans, 2023-06-29:
@UnexpectedMaker , I faced the same issues and simply use the classical option. 
This is probably the best option for low to mid-end hardware, see my video
https://youtu.be/B5dpPtdh3tc