30350n/pcb2blender #25: Ability to have the exporter ignore VIAs?

UnexpectedMaker, 2023-06-27:
I want to make some product shot renders of some of my boards, and though the via representation is accurate, I'd like to just not have them exported at all. 

Looking through the exporter code, it seems they are just coming from the PCB definition as plated holes due to their annulars, and I can't see a way to identify them to have them excluded.

Do you know if there is a way? Other than me having to hand modify the imported mesh in blender every time?

Thanks :)
Seon

30350n, 2023-06-27:
The pcb geometry is coming straight from KiCads VRML exporter, which the whole plugin builds upon. So there is no easy way to remove them ... without actually removing them in Blender. (The pad properties the exporter collects are only for automatic solder joint generation pretty much.)

There's some code on the [baking](https://github.com/30350n/pcb2blender/tree/baking) branch which iirc, does pretty much exactly this, but it'd probably needs some modification to be useful to you.

UnexpectedMaker, 2023-06-27:
Right - Ok, I'm asking on the KiCAD discord about how to go about getting VRML output without vias... if there's a way, or if the export process is written natively, or in an "export plugin" that can be hacked.

Probably not going to go anywhere, but worth a shot.

Thanks for the speedy replies!!!