30350n/pcb2blender #28: How to export glTF 2.0

N0cynym, 2023-07-05:
I just have a question:

How can I export from my blender project a `.glb` with rendered surface texture?

I want to create a file for https://modelviewer.dev/editor/

Thanks for this tool 🙏

30350n, 2023-07-06:
That's not easily possible atm, alteast not with the current Cycles shaders. The 3D (deprecated) material mode might work, but won't look nearly as good. Solutions for this have been discussed before in #8 and also #24 kinda.

N0cynym, 2023-07-07:
@30350n did you test how it looks like if you render top and bottom image and put it as image texture on the surface of the blank imported glb body?

30350n, 2023-07-07:
No, but what your suggesting is more or less a very naive version of [texture baking](https://docs.blender.org/manual/en/3.6/render/cycles/baking.html).
The problem is that this doesn't really work (or atleast isn't optimal) with the pcb mesh the addon produces. I spent some time on working around this on the [baking](https://github.com/30350n/pcb2blender/tree/baking) branch, but don't have time to further persue this atm.

N0cynym, 2024-01-19:
Dear @30350n,

I hope you started well into the new year!?

Do you have any news/updates about this feature?

30350n, 2024-01-19:
> I hope you started well into the new year!?

Yes, thanks, hope you did too.

> Do you have any news/updates about this feature?

No, not really and it wouldn't really be the main priority either. The main thing that needs to be done is a rework of all the materials, to improve performance (and visuals). Though this could also atleast simplify the export problem ...

Regardless, I can't really invest the time that would take atm.

N0cynym, 2024-01-19:
Thank you very much for your time and for updating this amazing project. 👍 

Well for now I'll see how I can embed a VRML export from KiCad to my web presentation. But I think using a glTF 2.0 export with your material finish in a [model viewer](https://modelviewer.dev/) would be much nicer ☺️

crazy-hair, 2024-07-31:
You can bake the textures out with the Blender passes (the ones you will need are Diffuse Color, Glossy Color, Roughness, Normal, and optionally Ambient Occlusion) onto a secondary UV map (you can use the integrated Smart UV Project function for this, it doesn't need to be fancy). Unfortunately Blender doesn't support exporting with the KHR_materials_pbrSpecularGlossiness extension though, so you'll have to export it as glTF and then edit the JSON with the proper specular setup:
[image]

If you want the solder mask to show up in the normal map, you'll need to bake the normal map pass (and AO, if you want it to show up there as well) with a diffuse node added to the original shader like this:
[image]
Don't bake the other passes like this or you'll get unwanted results.

Attached is an example glTF:
[glTF_compressed.zip](https://github.com/user-attachments/files/16436713/voltdiv_compressed.zip)

Make sure that the viewer you're using has support for SpecularGlossiness though, it looks like \<model-viewer\> might not support it. I know Babylon.js and PlayCanvas do.

TLDR: Not impossible, but certainly not a single button press. If you're familiar with Blender and the glTF standard it shouldn't be too difficult though.

crazy-hair, 2024-08-11:
Small update: It appears that with the Blender 4.2 update, you can export spec-gloss materials with a node setup like this: 
[image]
It works by setting the IOR to a very high index and then plugging the specular texture into the specular tint input. It actually doesn't utilize the `KHR_materials_pbrSpecularGlossiness` extension at all, instead just using `KHR_materials_ior` and `KHR_materials_specular` extensions, both of which appear to be supported by `<model-viewer>`.
[image]

With manual glTF editing out of the way, the only hurdle to exporting as glTF is now just baking the materials down to textures, and it's not that hard if you understand how rendering in Blender works. Here's an example glB file for reference: [compressed.zip](https://github.com/user-attachments/files/16571371/compressed.zip)

@N0cynym If you still need a pcb3d file converted to glTF, I can take a crack at it if you want to send it my way.

30350n, 2024-08-11:
Thanks a lot for looking into this and for sharing your findings! Ideally this would probably be automated, either by selecting a special "GLTF Export" material mode in the importer or by having a special "Export PCB to GLTF" operator somewhere, which could then also handle the baking as well as possibleGLTF post processing.

Regarding the "baking" part of this, I tried looking into this quite a while ago. My wip is available on the [`baking`](https://github.com/30350n/pcb2blender/tree/baking) branch.

crazy-hair, 2024-08-11:
It would probably be easiest to have a button labeled something like `Bake to exportable mesh` in the sidebar that clones the object, applies all the needed actions (make single user, apply modifiers, create a secondary UV map, auto-unwrap, etc,) and then bakes the materials to it. It would leave the user with a secondary mesh in the viewport that can be exported with the native glTF exporter. I unfortunately am not a dev, so I'm not really able to automate anything or make a PR, and I know you're busy with other things so I'm not expecting this to get implemented anytime soon, but on the face of it, it should be pretty doable.

hauserkristof, 2024-12-01:
Hey @crazy-hair , I would really be interested in a `Export PCB to glTF` "button" or something like that. I'm really new to Blender, I only want to have a simple but good looking render of my PCBs.

Mosaic-modular, 2024-12-02:
+1 for this functionality. I have been trying to do something similar with a missmatch of different plugins for blender but I never could get it to work.

hauserkristof, 2024-12-02:
@30350n do you plan to merge the master into `baking` or creating a tutorial maybe of the glTF export? 🤔 

Anyway, thank you for the awesome project!

hauserkristof, 2024-12-03:
If you are new to Blender (like me), there is a helpful short video about how to bake textures on a material:
https://www.youtube.com/watch?v=SDqpnfTRtIU

Based on this, and @crazy-hair 's Aug 11st shader workflow, it is not too hard to replicate and export to glTF successfully.

I hope this helps!

30350n, 2024-12-04:
> @30350n do you plan to merge the master into baking or creating a tutorial maybe of the glTF export? 🤔

Not really, as it's really WIP and not finished/ready at all.