30350n/pcb2blender #29: Importer error: AttributeError: 'NoneType' object has no attribute 'data'

Haschtl, 2023-09-23:
I can't import the pcb3d file in blender. It crashes with this output:
```
fake_module: addon missing 'bl_info' gives bad performance!: '/home/haschtl/.config/blender/3.6/scripts/addons/plugins/__init__.py'
gc.collect() -> 37
fake_module: addon missing 'bl_info' gives bad performance!: '/home/haschtl/.config/blender/3.6/scripts/addons/plugins/__init__.py'
Traceback (most recent call last):
  File "/usr/share/blender/3.6/scripts/addons/io_scene_x3d/import_x3d.py", line 1259, in gzipOpen
    filehandle = open(path, 'rU', encoding='utf-8', errors='surrogateescape')
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: invalid mode: 'rU'
Failed to open file: /tmp/pcb2blender_tmp/tester_pcb3d_16386390151959637076/pcb.wrl
Traceback (most recent call last):
  File "/usr/share/blender/3.6/scripts/addons/io_scene_x3d/import_x3d.py", line 1259, in gzipOpen
    filehandle = open(path, 'rU', encoding='utf-8', errors='surrogateescape')
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: invalid mode: 'rU'
Failed to open file: /tmp/pcb2blender_tmp/tester_pcb3d_16386390151959637076/components/DSUB-9_Female_Horizontal_P2.77x2.84mm_EdgePinOffset7.70mm_Housed_MountingHolesOffset9.12mm.wrl
Traceback (most recent call last):
  File "/usr/share/blender/3.6/scripts/addons/io_scene_x3d/import_x3d.py", line 1259, in gzipOpen
    filehandle = open(path, 'rU', encoding='utf-8', errors='surrogateescape')
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: invalid mode: 'rU'
Failed to open file: /tmp/pcb2blender_tmp/tester_pcb3d_16386390151959637076/components/Altech_AK300_1x02_P5.00mm_45-Degree.wrl
Traceback (most recent call last):
  File "/home/haschtl/.config/blender/3.6/scripts/addons/pcb2blender_importer/importer.py", line 154, in execute
    if (pcb := self.import_pcb3d(context, filepath)) == {"CANCELLED"}:
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/haschtl/.config/blender/3.6/scripts/addons/pcb2blender_importer/importer.py", line 297, in import_pcb3d
    obj.data.name = component.rsplit("/", 1)[1].rsplit(".", 1)[0]
    ^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'data'
```

I'm trying to load some demo project of KiCad 7.  

OS: Linux (Arch)
pcb2blender version: 2.6
Blender version: 3.6.2
KiCad version: 7.0.7

30350n, 2023-09-23:
This is not an issue with this addon, but with the wrl/x3d importer, though one which has already been fixed for a while now. There's something weird about your installation that code should not be part of any recent Blender build/install.

30350n, 2023-09-23:
For reference, this got fixed 5 months ago: https://github.com/blender/blender-addons/blame/57cdb9e186ead9b34e5b91c1e2665661cfc67138/io_scene_x3d/import_x3d.py#L1261

Haschtl, 2023-09-23:
Strange...
I replaced the import_x3d.py file with the upstream version and now it works on my local computer.
Nevertheless, I'm trying to integrate this process in CI/CD. pcb2blender is (almost) integrated in KiBot, only the CI part is missing. So I extended the kibot-docker container with blender and the pcb2blender-importer, but this Image is facing the same error. Here is the corresponding issue: INTI-CMNB/KiBot/issues/499

30350n, 2023-09-23:
How are you installing Blender? Got a docker file to reference somewhere?

Haschtl, 2023-09-23:
I posted the Dockerfile in the referenced ticket:

```
FROM ghcr.io/inti-cmnb/kicad7_auto_full:latest

COPY installPcb2blenderImporter.py ./installPcb2blenderImporter.py
RUN apt update
RUN pip3 install skia-python --break-system-packages
RUN apt install -y blender wget python3-pil python3-numpy
RUN wget https://github.com/30350n/pcb2blender/releases/download/v2.6-k7.0-b3.5/pcb2blender_importer_2-6.zip
RUN blender -b  -y  --python ./installPcb2blenderImporter.py

WORKDIR /mnt
```

Haschtl, 2023-09-23:
You can find the full repo building the docker image here: https://github.com/Haschtl/kicad7_auto_full_blender

30350n, 2023-09-23:
Are you sure the blender version is 3.6.x? From what I can tell, all these docker images are based on [kicad_debian](https://github.com/INTI-CMNB/kicad_debian/tree/master) (though it's hard to see through this docker container dependency hell). There's two images in there that install blender, the [`base_os_full`](https://github.com/INTI-CMNB/kicad_debian/tree/master/base_os_full) and [`base_os_nightly_full`](https://github.com/INTI-CMNB/kicad_debian/tree/master/base_os_nightly_full) ones. The first one installs Blender 3.5 from [here](https://mirrors.ocf.berkeley.edu/blender/release/Blender3.5/blender-3.5.1-linux-x64.tar.xz) and that one seems fine (no `"rU"`). The `nightly` one (for whatever reason) still installs Blender 3.4 though, from [here](https://mirrors.ocf.berkeley.edu/blender/release/Blender3.4/blender-3.4.1-linux-x64.tar.xz) and that one indeed still contains the code using the deprecated `"rU"` file mode.

30350n, 2023-09-23:
So I guess, for whatever reason you are ending up with that old Blender 3.4 installation.

Haschtl, 2023-09-23:
Good point! 
I didn't see, that the kicad_debian image already installs Blender 3.5.1 and I overwrote it with `apt install blender` to version 3.4

I removed the line from my Dockerfile and now it works great. Here is the [working docker image](https://github.com/Haschtl/kicad7_auto_full_blender/pkgs/container/kicad7_auto_full_blender)

closed 2023-09-23 by Haschtl

30350n, 2023-09-23:
Ah that makes sense 👍
Actually ... looking at that docker file you created, nothing in there should be required really. The python dependencies aswell as the addon also already get installed by the docker files.

Haschtl, 2023-09-23:
Good point! I've looked at the Dockerfile of KiBot, indeed the plugin is beeing installed. 
May the problem be, that it's using pcb2blender v2.1, which does not support KiCad7 and blender >=3.5 ?
Here is the [Dockerfile](https://github.com/INTI-CMNB/KiBot/blob/1af4204def73bcb1c62aba5d11ab9872c173bddb/experiments/blender/create_image/Dockerfile#L22), i'm referencing to.

30350n, 2023-09-23:
Not sure where that one is coming from, I'm talking about the [kicad_auto_test](https://github.com/INTI-CMNB/kicad_auto_test/tree/master) ones, specifically [this](https://github.com/INTI-CMNB/kicad_auto_test/blob/master/ki7/Dockerfile) one (which installs 2.6).

Haschtl, 2023-09-23:
Oh yes, you are right! Indeed the current docker container of KiBot includes everything needed!
I was able to generate the .blend file and the other outputs for the test project using the kicad7_auto_full:latest image.
I'm still facing another issue with my real project, but I will create another issue on that topic (very large PCB)