30350n/pcb2blender #30: Blender-Import doesn't work for large PCBs (DecompressionBombWarning)

Haschtl, 2023-09-23:
The Blender importer crashes if the PCB is very large (315mm x 315mm) :

```
DEBUG:- Output from command: /usr/bin/3.5/python/lib/python3.10/site-packages/PIL/Image.py:3157: DecompressionBombWarning: Image size (166306816 pixels) exceeds limit of 89478485 pixels, could be decompression bomb DOS attack.
  warnings.warn(
Blender 3.5.1 (hash e1ccd9d4a1d3 built 2023-04-24 23:31:15)
Traceback (most recent call last):
  File "/usr/bin/3.5/scripts/addons/pcb2blender_importer/importer.py", line 952, in execute
    result = import_x3d.load(context, self.filepath, global_matrix=matrix)
  File "/usr/bin/3.5/scripts/addons/io_scene_x3d/import_x3d.py", line 3635, in load
    load_web3d(context, filepath,
  File "/usr/bin/3.5/scripts/addons/io_scene_x3d/import_x3d.py", line 3534, in load_web3d
    importShape(bpycollection, node, ancestry, global_matrix)
  File "/usr/bin/3.5/scripts/addons/io_scene_x3d/import_x3d.py", line [314](https://git.cre.science/product-hardware/fluxshield-mk-2/-/jobs/10374#L314)8, in importShape
    bpydata = geom_fn(geom, ancestry)
  File "/usr/bin/3.5/scripts/addons/io_scene_x3d/import_x3d.py", line 1900, in importMesh_IndexedFaceSet
    if len(points) >= 2 * len(index):  # Need to cull
TypeError: object of type 'NoneType' has no len()
Traceback (most recent call last):
  File "/usr/bin/3.5/scripts/addons/pcb2blender_importer/importer.py", line 154, in execute
    if (pcb := self.import_pcb3d(context, filepath)) == {"CANCELLED"}:
  File "/usr/bin/3.5/scripts/addons/pcb2blender_importer/importer.py", line 294, in import_pcb3d
    bpy.ops.pcb2blender.import_x3d(
  File "/usr/bin/3.5/scripts/modules/bpy/ops.py", line 113, in __call__
    ret = _op_call(self.idname_py(), None, kw)
RuntimeError: Error: Python: Traceback (most recent call last):
  File "/usr/bin/3.5/scripts/addons/pcb2blender_importer/importer.py", line 952, in execute
    result = import_x3d.load(context, self.filepath, global_matrix=matrix)
  File "/usr/bin/3.5/scripts/addons/io_scene_x3d/import_x3d.py", line 3635, in load
    load_web3d(context, filepath,
  File "/usr/bin/3.5/scripts/addons/io_scene_x3d/import_x3d.py", line 3534, in load_web3d
    importShape(bpycollection, node, ancestry, global_matrix)
  File "/usr/bin/3.5/scripts/addons/io_scene_x3d/import_x3d.py", line 3148, in importShape
    bpydata = geom_fn(geom, ancestry)
  File "/usr/bin/3.5/scripts/addons/io_scene_x3d/import_x3d.py", line 1900, in importMesh_IndexedFaceSet
    if len(points) >= 2 * len(index):  # Need to cull
TypeError: object of type 'NoneType' has no len()
Location: /usr/bin/3.5/scripts/modules/bpy/ops.py:113
```

The Pillow library has a variable to extend this limit, e.g. `Image.MAX_IMAGE_PIXELS = 933120000` ([StackExchange](https://stackoverflow.com/questions/51152059/pillow-in-python-wont-let-me-open-image-exceeds-limit))
I think, this would fix problems with large PCBs

30350n, 2023-09-23:
The pillow thing is only a warning though (which might or might no cause issues later), the main error is from the x3d_importer again.

30350n, 2023-09-23:
Can you share the .pcb3d file?

Haschtl, 2023-09-23:
I had (partially) broken wrl files in a custom kicad-library, which caused this issue. It seems WRL files from FreeCAD can sometimes not be imported in blender. Importing the wrl in Meshlab and exporting it as wrl again, solves these broken wrl files.

closed 2023-09-23 by Haschtl

set-soft, 2023-09-24:
Hi @Haschtl !

I had the same problem using some old WRLs exported by a colleague (from STEP files). Lamentably the Blender X3D importer doesn't do a good sanity check.