30350n/pcb2blender #31: Blender-Importer error with some boards: edges.new(): this edge exists

Haschtl, 2023-09-23:
I encountered an issue with the `improve_board_mesh` function: `edges.new(): this edge exists` in the statement `bm.edges.new((top_vert, bot_vert))`.

I think, this error can occur for boards with complex outlines, that may be not defined properly.

In my case it works with a simple try/except:
```python
            try:
                bm.edges.new((top_vert, bot_vert))
            except ValueError as e:
                print(e)
```

30350n, 2023-09-23:
That's a bad solution imo, especially because I already got a check which *should* catch this right above [here](https://github.com/30350n/pcb2blender/blob/master/pcb2blender_importer/importer.py#L744). So this shouldn't really happen. Though iirc I'm kind of abusing the way KiCad writes the PCB mesh with that code, so if they ever change something about that it might not work anymore. Can you share the pcb3d file for this one? 😅

Haschtl, 2023-09-23:
Yes, not the best solution... how about this:
```
            existing=bm.edges.get((top_vert, bot_vert))
            if existing is None:
                bm.edges.new((top_vert, bot_vert))
```

Unfortunately, I cannot share this pcb3d file :(

30350n, 2023-09-23:
Can you share it privately? (There's a mailto link at the bottom of [my website](https://30350n.de/)). Also, the `pcb.wrl` file from the .pcb3d file would be enough probably (.pcb3d files are just .zip files so you should be able to extract it quite easily).

30350n, 2023-09-24:
Should be fixed in [v2.7](https://github.com/30350n/pcb2blender/releases/tag/v2.7-k7.0-b3.6) now!

30350n, 2023-10-04:
Had any time to test this again? @Haschtl

closed 2024-01-03 by 30350n