30350n/pcb2blender #32: Adding The Blender doesn't work

o7-machinehum, 2023-10-05:
I'm following all the steps; the output looks like this.

```
Modules Installed () from '/Users/machinehum/Downloads/pcb2blender_exporter_2-7.zip' into '/Users/machinehum/Library/Application Support/Blender/3.6/scripts/addons'
```

The addon isn't showing up in the list. It's a Macbook Pro M2 running Blender 3.6.4

30350n, 2023-10-05:
Can you share a screenshot of the addon not being there in the user prefs? Also, try starting blender from a terminal and check if there's any error appearing in there (if so, please screenshot them aswell).

30350n, 2023-10-12:
Actually, on a second glance: You are trying to install the pcb2blender_**exporter** Kicad plugin into Blender, which obviously doesn't work. Try to install the pcb2blender_**importer**.

closed 2023-10-12 by 30350n