30350n/pcb2blender #35: Export error with the latest 2.7 release

ojthetiny, 2023-10-08:
Updated my plugin to the new 2.7 version and now I can't export the file out of kicad, error message seems to be permission related despite running kicad as admin
[image]

30350n, 2023-10-08:
Hm, that is weird. For some reason you don't seem to have permission to delete the `pcb2blender_tmp` folder. Try deleting it manually from your file explorer (only the `pcb2blender_tmp` part, **not** the whole `Local\Temp` folder). That should fix it.
You should never be required to have admin permissions to use the addon btw.

ojthetiny, 2023-10-08:
yes, that did the trick, thank you

closed 2023-10-12 by 30350n