30350n/pcb2blender #36: Problems using custom solder mask colors

set-soft, 2023-10-11:
I'm having problems testing the case where a color is defined using its hexadecimal representatio, the following code:

https://github.com/30350n/pcb2blender/blob/05d53b933a5deba76b6c9e8522932f97f8587d8a/pcb2blender_importer/materials.py#L87

Produces *TypeError: can't multiply sequence by non-int of type 'float'*

Which is consistent with what I get in the interpreter:

```python
>>> color = (1,2,3)
>>> (color * 0.2)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: can't multiply sequence by non-int of type 'float'
```

Perhaps you can use something like:

```python
>>> list(map(lambda x: x*0.2, color))
[0.2, 0.4, 0.6000000000000001]
```

set-soft, 2023-10-12:
> Even better, I can just use Blenders builtin `Color` type. Thanks for the catch, should be fixed now!

Yes, this seems to be ok. But isn't enough to solve the problems. After applying it I'm getting:

```
RuntimeError: Error: Python: Traceback (most recent call last):
  File "/usr/bin/3.5/scripts/addons/pcb2blender_importer/importer.py", line 205, in execute
    if (pcb := self.import_pcb3d(context, filepath)) == {"CANCELLED"}:
  File "/usr/bin/3.5/scripts/addons/pcb2blender_importer/importer.py", line 375, in import_pcb3d
    setup_pcb_material(board_material.node_tree, images, pcb.stackup)
  File "/usr/bin/3.5/scripts/addons/pcb2blender_importer/materials.py", line 144, in setup_pcb_material
    setup_node_tree(node_tree, nodes, label_nodes=False)
  File "/usr/bin/3.5/scripts/addons/pcb2blender_importer/custom_node_utils/__init__.py", line 19, in setup_node_tree
    setattr(node, attr, value)
TypeError: bpy_struct: item.attr = val: enum "CUSTOM" not found in ('GREEN', 'RED', 'YELLOW', 'BLUE', 'PURPLE', 'WHITE', 'BLACK', 'MATTE_BLACK')
Location: /usr/bin/3.5/scripts/modules/bpy/ops.py:113
```

30350n, 2023-10-12:
Ah yeah okay, I see I messed up there, should also be fixed now 🤞

set-soft, 2023-10-18:
Hi @30350n !

Close, I had to also change:

```diff
--- materials.py	2023-10-18 13:01:04.874202150 -0300
+++ /tmp/materials.py	2023-10-18 13:39:41.123547271 -0300
@@ -239,7 +239,6 @@
     bl_width_default = 180
 
     def update_props(self, context):
-        light_color, dark_color = MASK_COLOR_MAP[self.soldermask]
         match self.soldermask:
             case "WHITE":
                 roughness = 0.15
@@ -249,6 +248,7 @@
                 roughness = 0.25
 
         if self.soldermask != "CUSTOM":
+            light_color, dark_color = MASK_COLOR_MAP[self.soldermask]
             self.inputs["Light Color"].default_value = (*srgb2lin(light_color), 1.0)
             self.inputs["Dark Color"].default_value  = (*srgb2lin(dark_color),  1.0)
             self.inputs["Roughness"].default_value = roughness
```

30350n, 2023-10-18:
Should now actually be fixed. Didn't really have time to properly test this yet until now, sorry. I've added a test file now and everything passes, so this should be fine now.

set-soft, 2023-10-19:
Perfect, I tried the last code and it works. I defined some "pink" color for the @mundodisco8 snow flake board (the one that triggered upgrading to 2.7 to get correct solder mask color) and got:

[image]

closed 2023-10-19 by set-soft

mundodisco8, 2023-10-20:
Damn! now I want to make a run with pink soldermask 🤣