30350n/pcb2blender #37: How to render PTH?

VIPQualityPost, 2023-10-19:
My PTH always are showing as filled with solder, which is OK in some cases (like unpopulated THT parts) but for others (like mounting holes) this is not desired. Is there a way to indicate which holes should be filled? or is this an issue with the footprints? I am using the default KiCad "pad" PTH mounting hole footprint.
Example, the corner holes here: 
[image]

VIPQualityPost, 2023-10-19:
oh, they just get filled with a solder blob object. you can highlight it in blender and just delete it!

closed 2023-10-19 by VIPQualityPost

30350n, 2023-10-20:
Hmmm this seems weird, normally mounting holes, unpopulated components etc. shouldn't get those solder blobs. Do you have the "Add Solder Joints" setting set to "Smart"?
[image]

VIPQualityPost, 2023-10-20:
I didn't try tinkering with those settings yet - let me see.

30350n, 2023-10-20:
Hmmm, well it should be set to "Smart" by default. If it still includes the solder blobs with that setting, could you double check what footprints you are using (mainly for the mounting holes I guess) and possibly also share the .pcb3d file?

VIPQualityPost, 2023-10-20:
Should the mounting holes be marked "do not populate"? I suppose so, right? I haven't set that. 
It is set to "smart".

30350n, 2023-10-20:
They shouldn't be drawn if they are DNP, yeah (though that'd make more sense for components which *could* actually be assembled). For mounting holes, the Pad should be set to NPTH, Mechanical.
[image]

(There's a few more conditions like, there won't be any solder if no 3D model is defined for example).