30350n/pcb2blender #40: Connector rotation incorrect

oliv3r, 2023-12-17:
I'm not sure if this is an issue with pcb2blender, or if the issue is elsewhere, so apologies if I'm reporting it in the wrong location.

My workflow is currently using kibot's docker kicad 7 full image, which should have all the tools installed. The configuration snippet I use,
```yaml
  - name: '3D_top_view'
    comment: '3D render from the top.'
    type: 'blender_export'
    dir: 'output/blender/'
    options:
      render_options:
        transparent_background: false
        samples: 1
      point_of_view:
        view: 'top'
      outputs:
        - type: 'blender'
        - type: 'render'
```

I made a PCB, which has a 3D file (from lcsc) but is abused ('its a through hole part, that I rendered rotated by 90 degrees to make it an edge connector). Also the lcsc part was offset, on the Z (-2.2) axis to compensate for the height, the X (-2.5) to get it from half the location to the correct location, and the Y (30.5) to move the pads in the correct spot. All in all, a few adjustment, but nothing major, and probably not relevant. (p.s. I realize that my story's coordinates don't match the offsets, but I'm sure you understand what I mean :p)

Opening the generated blender file, I see that the X and Z rotation are at -43.06 and 43.06. This is obviously wrong, which results in the following image.
[image]

Adjusting the X and Y rotations to -45 and +45, everything is perfect. So where do these rotations come from? Even though I'm abusing the wrong part, all modifications I made cause no problems in kicad itself. And the fact that both X and Z are incorrect by the same amount, and everything else is perfect, is surprising to me. I've attached the pcb3d (renamed to pcb3d.zip) but I'm not sure if all the needed information is in there. Who's responsible for mapping these transformations? Everything else renders really well.

[breakout_hdmi-blender_export.pcb3d.zip](https://github.com/30350n/pcb2blender/files/13694605/breakout_hdmi-blender_export.pcb3d.zip)

30350n, 2023-12-17:
Hmm, that definitely seems like something weird is going on. Did some quick tests with your provided `.pcb3d` file and the issue also occurs if I just import the `pcb.wrl` file. So the issue is either present in both, Blender's VRML importer **and** my code, or it's on the export side (i.e. KiCads VRML exporter), which I find somewhat more likely.

Can you share the KiCad pcb file you are exporting from? Also, what KiCad version are you using exactly?

oliv3r, 2023-12-17:
> Hmm, that definitely seems like something weird is going on. Did some quick tests with your provided `.pcb3d` file and the issue also occurs if I just import the `pcb.wrl` file. So the issue is [...]

I'm using the kibot docker container, ghcr.io/inti-cmnb/kicad7_auto_full:latest with the digest 20587c39f16b93ab90962f09add23886a92152f1a6e76a206461e881565d2c0f

<details>
<summary>kibot info output</summary>

```
KiBot installation checker

Core:
Linux: 6.6.7 (Linux 11a9c891eecc 6.6.7-arch1-1 #1 SMP PREEMPT_DYNAMIC Thu, 14 Dec 2023 03:45:42 +0000 x86_64 GNU/Linux)
  /usr/bin/uname
Python: 3.11.2 (main, Mar 13 2023, 12:18:29) [GCC 12.2.0]
  /usr/lib/python3.11/os.py
  /usr/bin/python3
KiCad: 7.0.9-7.0.9~ubuntu23.04.1
  /usr/lib/python3/dist-packages/pcbnew.py
  /usr/bin/kicad
Kibot: 1.6.3
  /usr/lib/python3/dist-packages/kibot/__main__.py
  /usr/bin/kibot

Modules:
Colorama: 0.4.6
  /usr/lib/python3/dist-packages/colorama/__init__.py
LXML: 4.9.2
  /usr/lib/python3/dist-packages/lxml/__init__.py
Lark: 1.1.5
  /usr/lib/python3/dist-packages/lark/__init__.py
PyYAML: 6.0.0 (6.0)
  /usr/lib/python3/dist-packages/yaml/__init__.py
QRCodeGen: Ok
  /usr/lib/python3/dist-packages/qrcodegen.py
Requests: 2.28.1
  /usr/lib/python3/dist-packages/requests/__init__.py
XLSXWriter: 3.0.2
  /usr/lib/python3/dist-packages/xlsxwriter/__init__.py
Xvfbwrapper: Ok
  /usr/lib/python3/dist-packages/xvfbwrapper.py
markdown2: 2.4.1
  /usr/lib/python3/dist-packages/markdown2.py
mistune: 2.0.4
  /usr/lib/python3/dist-packages/mistune/__init__.py
numpy: 1.24.2
  /usr/lib/python3/dist-packages/numpy/__init__.py

Tools:
Bash: 5.2.15 (GNU bash, version 5.2.15(1)-release (x86_64-pc-linux-gnu))
  /usr/bin/bash
Blender: 3.5.1 (Blender 3.5.1)
  /usr/bin/blender
Ghostscript: 10.0.0 (10.00.0)
  /usr/bin/gs
Git: 2.39.2 (git version 2.39.2)
  /usr/bin/git
ImageMagick: 6.9.11.60 (Version: ImageMagick 6.9.11-60 Q16 x86_64 2021-01-25 https://imagemagick.org)
  /usr/bin/convert
Interactive HTML BoM: 2.7.0 (v2.7.0)
  /usr/bin/python3
KiBoM: 1.9.1 (KiBOM Version: 1.9.1)
  /usr/bin/KiBOM_CLI.py
KiCad Automation tools (kiauto): 2.2.8 (pcbnew_do 2.2.8 - Copyright 2018-2023, INTI/Productize SPRL - License: Apache)
  /usr/bin/pcbnew_do
KiCad PCB/SCH Diff (kidiff): 2.4.7 (kicad-diff.py 2.4.7 - Copyright 2020-2023, INTI/Salvador E. Tropea - License:)
  /usr/bin/kicad-diff.py
KiCost: 1.1.18 (KiCost v1.1.18)
  /usr/bin/kicost
KiKit: 1.4.0.1 (kikit, version 1.4.0-1)
  /usr/bin/kikit
OpenSCAD: 2021.1.0 (OpenSCAD version 2021.01)
  /usr/bin/openscad
Pandoc: 2.17.1.1 (pandoc 2.17.1.1)
  /usr/bin/pandoc
RAR: 6.20.0 (RAR 6.20   Copyright (c) 1993-2023 Alexander Roshal   17 Jan 2023)
  /usr/bin/rar
RSVG tools: 2.54.5 (rsvg-convert version 2.54.5)
  /usr/bin/rsvg-convert
Xvfb: Ok (xvfb-run)
  /usr/bin/rsvg-convert

Environment:
HOME
HOSTNAME
INTERACTIVE_HTML_BOM_NO_DISPLAY
KIBOT_ITERATION
KICAD_AUTO_FULL
LC_CTYPE
OLDPWD
PATH
PWD
SHLVL
TERM
TZ
_

KiAuto:
This is KiAuto v2.2.8
Installed at: /usr/bin/pcbnew_do
Using kiauto module from: /usr/lib/python3/dist-packages/kiauto
Interpreted by Python: /usr/bin/python3 (v3.11.2 (main, Mar 13 2023, 12:18:29) [GCC 12.2.0])
Tools:
- kicad: /usr/bin/kicad (v7.0.9-7.0.9~ubuntu23.04.1)
- xdotool: /usr/bin/xdotool
- recordmydesktop: /usr/bin/recordmydesktop
- xsltproc: /usr/bin/xsltproc
- xclip: /usr/bin/xclip
- convert: /usr/bin/convert
```
</details>

[breakout_hdmi.kicad_pcb.zip](https://github.com/30350n/pcb2blender/files/13696713/breakout_hdmi.kicad_pcb.zip)

set-soft, 2023-12-18:
Hi @oliv3r and @30350n !

I loaded the WRL using FreeCAD and it looks ok: 
[image]

I'll suggest using stepUp to make the 3D model usable without any offset and/or rotation. Even when these values are available KiCad behaves very poorly honoring them when exporting to VRML and STEP formats.

30350n, 2023-12-18:
Huh, odd! Guess it's an issue with Blender's VRML importer/my code after all then, will check that in the next few days ^^

30350n, 2023-12-18:
> I'll suggest using stepUp to make the 3D model usable without any offset and/or rotation.

I'd suggest [free2ki](https://github.com/30350n/free2ki) :P

oliv3r, 2023-12-18:
I'd rather not 'fix the wrl' as it's something I just got from the kicad library/lcsc library. That said, if we got a bug somewhere, then there's a bug somewhere that ought to be fixed?

30350n, 2023-12-19:
Yep, there's probably a bug somewhere in the code that handles VRML rotations in Blender (as well as in my addon, probably because I more or less copied that code). The .wrl files are fine, nothing to fix about them.

set-soft, 2023-12-20:
> I'd rather not 'fix the wrl' as it's something I just got from the kicad library/lcsc library. 

I disagree with @30350n. If the WRL needs rotations and/or offsets it must be fixed. KiCad can't really handle it.
You might be using a 3D model from KiCad libs, but you are tweaking its use, so you should rename it and adjust it for your needs. Otherwise you'll see problems in various places.

oliv3r, 2023-12-20:
> I disagree with @30350n. If the WRL needs rotations and/or offsets it must be fixed. KiCad can't really handle it. You might be using a 3D model from KiCad libs, but you are tweaking its use, so you [...]

I"m not really. I'm only rotating it a bit. Nothing that matches that weird diagonal offset even.

30350n, 2023-12-20:
> I disagree with @30350n. If the WRL needs rotations and/or offsets it must be fixed. KiCad can't really handle it. You might be using a 3D model from KiCad libs, but you are tweaking its use, so you [...]

Why would KiCad not be able to handle simple transformations? 🤔 
I'm translating/rotating models in KiCad all the time and never had an issue with that. Also in this case the exported .wrl looks fine in FreeCad but not in Blender, so KiCad is obviously not the issue here.

set-soft, 2023-12-20:
> Why would KiCad not be able to handle simple transformations? 🤔 I'm translating/rotating models in KiCad all the time and never had an issue with that. Also in this case the exported .wrl looks fine in FreeCad but not in Blender, so KiCad is obviously not the issue here.

In years using KiCad I found a lot of issues with this, until I started using stepUp to avoid offsets and rotations. No more problems.

30350n, 2023-12-20:
Well I've only used it since 5.9 only so maybe it has only recently been working flawlessly 🤷 
StepUp still seems like one big but though, pretty much unusable software imo.

30350n, 2024-01-05:
Great so this seems to be a [fundamental bug in Blender's Euler conversion code](https://projects.blender.org/blender/blender/issues/96251#issuecomment-1095862) which got reported 2 years ago, but noone could be bothered to fix it.

oliv3r, 2024-01-05:
@30350n ouch, well nice that you've found something about it!! That's step 1 :D

30350n, 2024-01-09:
Cool, just tried the newest Alpha 4.1 build from https://builder.blender.org and it seems like they [fixed it](https://projects.blender.org/blender/blender/issues/116880)!
[image]

oliv3r, 2024-01-16:
Amazing! Why does your render look better then mine? :p anyway, I'd be happy to close this issue, we can always re-open it if after 4.1 lands and this still isn't resolved.

@set-soft I suppose once 4.1 is released we can include it in 1.6.4 or 1.6.5?

set-soft, 2024-01-16:
> @set-soft I suppose once 4.1 is released we can include it in 1.6.4 or 1.6.5?

Blender 4.1 will be stable on march, here is the [schedule](https://projects.blender.org/blender/blender/milestone/18) 

I think this will be for 1.6.5.

30350n, 2024-01-17:
> Amazing! Why does your render look better then mine? :p

Lighting probably 😅 . I'm mainly using HDRIs from [polyhaven](https://polyhaven.com/) with their [Gaffer Addon](https://github.com/gregzaal/Gaffer/). To be exact, here I used [this one](https://polyhaven.com/a/abandoned_factory_canteen_02) ^^

> anyway, I'd be happy to close this issue

sounds good!

closed 2024-01-21 by 30350n

oliv3r, 2024-01-21:
@set-soft can we do this lightening plugin in kibot too? :D

set-soft, 2024-01-22:
Hi @oliv3r!

I don't know what exactly is needed. If you can try it and explain it step by step I could try it, what is needed? how it works?

30350n, 2024-01-22:
You'd have to:
1. Install the [Gaffer](https://github.com/gregzaal/Gaffer/) addon
2. Create an hdri folder and add it via `bpy.ops.gaffer.hdri_path_add(...)` in Blender
3. Offer an option to set a hdri from polyhaven, so that `abandoned_factory_canteen_02` would for example download the hdri from `https://dl.polyhaven.org/file/ph-assets/HDRIs/exr/4k/abandoned_factory_canteen_02_4k.exr` to the hdri folder.
4. Enable the Gaffer HDRI handler via `bpy.context.scene.gaf_props.hdri_handler_enabled = True`
5. Select the HDRI via `gaf_props.hdri_handler_enabled = <path to hdri file>`

Depending on how 5. works exactly, 2. could maybe even be skipped.

30350n, 2024-01-22:
> > 1. Install the [Gaffer](https://github.com/gregzaal/Gaffer/) addon
> 
> I tried copying it to the scripts dir and then tried:
> 
> import addon_utils addon_utils.enable("Gaffer")
> 
> It fails:
> 
> ```
> Traceback (most recent call last):
>   File "/usr/bin/3.5/scripts/modules/addon_utils.py", line 369, in enable
>     mod.register()
>   File "/home/salvador/.config/blender/3.5/scripts/addons/Gaffer/__init__.py", line 710, in register
>     bpy.context.preferences.addons[__name__].preferences, bpy.context
> KeyError: 'bpy_prop_collection[key]: key "Gaffer" not found'
> ```
> 
> Looks like the code is wrong and works only when already registered. Which may be true when using the GUI.

You have to use the name of the folder inside of the zip.
[image]

30350n, 2024-01-22:
> > 2. Create an hdri folder and add it via `bpy.ops.gaffer.hdri_path_add(...)` in Blender
> 
> This also fails, looks like this is interactive:
> 
> ```
> Traceback (most recent call last):
>   File "<blender_console>", line 1, in <module>
>   File "/usr/bin/3.5/scripts/modules/bpy/ops.py", line 111, in __call__
>     ret = _op_call(self.idname_py(), C_dict, kw, C_exec, C_undo)
> TypeError: Calling operator "bpy.ops.gaffer.hdri_path_add" error, expected a string enum in ('INVOKE_DEFAULT', 'INVOKE_REGION_WIN', 'INVOKE_REGION_CHANNELS', 'INVOKE_REGION_PREVIEW', 'INVOKE_AREA', 'INVOKE_SCREEN', 'EXEC_DEFAULT', 'EXEC_REGION_WIN', 'EXEC_REGION_CHANNELS', 'EXEC_REGION_PREVIEW', 'EXEC_AREA', 'EXEC_SCREEN')
> ```

`bpy.ops.gaffer.hdri_path_add(directory=...)` should work.

set-soft, 2024-01-22:
> `bpy.ops.gaffer.hdri_path_add(directory=...)` should work.

Seems to work!

set-soft, 2024-01-23:
> What did you use? 🤔

I cloned the repo, so the dir is named "Gaffer". Blender 3.5.1 finds it, but the mechanism used to "register" the preferences:

```python
ui.update_category(bpy.context.preferences.addons[__name__].preferences, bpy.context)
```

From the *register* function fails when programmatically enabling the add-on. It works if I use the GUI, but this is impossible for the CI/CD. I'm enabling *pcb2blender* using the same code:

```python
import addon_utils
addon_utils.enable("pcb2blender_importer")
```

30350n, 2024-01-23:
How does it "fail" exactly?

set-soft, 2024-01-23:
> How does it "fail" exactly?

```
Traceback (most recent call last):
  File "/usr/bin/3.5/scripts/modules/addon_utils.py", line 369, in enable
    mod.register()
  File "/home/salvador/.config/blender/3.5/scripts/addons/Gaffer/__init__.py", line 710, in register
    bpy.context.preferences.addons[__name__].preferences, bpy.context
KeyError: 'bpy_prop_collection[key]: key "Gaffer" not found'
```

I don't know much about the registration process, addon_utils.modules() finds the module and, as the traceback shows, the module is loaded and the *register* function called. Don't know what is missing ... looks like some registration in the *bpy_prop_collection*

30350n, 2024-01-23:
Hmm, odd. Do you possibly get any errors *before* that? I could imagine it failing sooner (pre-registration), thus the addon preferences wouldn't be created correctly.

set-soft, 2024-01-23:
> Hmm, odd. Do you possibly get any errors _before_ that? I could imagine it failing sooner (pre-registration), thus the addon preferences wouldn't be created correctly.

Nope, this is the first error I get when I start from scratch