30350n/pcb2blender #41: No pads or silk

bretskee, 2024-01-04:
using blender 4.0 and kicad 7.0.1. Pads and silk screens do not appear. Dont know if this is export or import issue (or just finger trouble)

30350n, 2024-01-04:
Whst OS are you on?

bretskee, 2024-01-04:
Windows 11 Version 22H2 (OS Buikd 22621.2861)

30350n, 2024-01-04:
Ok, try to open the System Console in Blender (`Window -> Toggle System Console`) before importing. See if any errors come up in there. You can also share the `.pcb3d` file if you want.

bretskee, 2024-01-05:
The console is silent. My .pcb3d file is attached renamed as .txt.

[EXPORT.txt](https://github.com/30350n/pcb2blender/files/13847128/EXPORT.txt)

30350n, 2024-01-06:
Looks good to me. You just have to go into rendered mode, guess you didn't do that?
[image]

bretskee, 2024-01-06:
Guess not... thanks.

closed 2024-01-06 by bretskee