30350n/pcb2blender #42: key "position" not found

rmd6502, 2024-01-04:
Created a text item for TL and BR as specified, in the "User Comments" layer.
Went into blender 3.4.1 and selected "import pcb3d"
Got the following:
`
﻿﻿bpy.context.space_data.system_bookmarks_active = 1
bpy.ops.object.select_all(action='DESELECT')
bpy.ops.object.shade_smooth()
bpy.ops.object.transform_apply(location=False, rotation=False)
bpy.ops.object.editmode_toggle()
bpy.ops.mesh.select_all(action='SELECT')
bpy.ops.mesh.tris_convert_to_quads()
bpy.ops.object.editmode_toggle()
Python: Traceback (most recent call last):
  File "/Users/rmd6502/Library/Application Support/Blender/3.4/scripts/addons/pcb2blender_importer/importer.py", line 205, in execute
    if (pcb := self.import_pcb3d(context, filepath)) == {"CANCELLED"}:
  File "/Users/rmd6502/Library/Application Support/Blender/3.4/scripts/addons/pcb2blender_importer/importer.py", line 313, in import_pcb3d
    self.setup_uvs(obj, pcb.layers_bounds)
  File "/Users/rmd6502/Library/Application Support/Blender/3.4/scripts/addons/pcb2blender_importer/importer.py", line 711, in setup_uvs
    mesh.attributes["position"].data.foreach_get("vector", vertices)
KeyError: 'bpy_prop_collection[key]: key "position" not found'
`
[class 2a.kicad_pcb.zip](https://github.com/30350n/pcb2blender/files/13833959/class.2a.kicad_pcb.zip)

rmd6502, 2024-01-04:
[heart.pcb3d.zip](https://github.com/30350n/pcb2blender/files/13833968/heart.pcb3d.zip)

30350n, 2024-01-04:
What addon version are you using? You seem to be using Blender 3.4, which is the problem here.

rmd6502, 2024-01-04:
The latest - using my phone for internet right now, will upgrade blender when on wifi again

30350n, 2024-01-04:
For the latest version you'll need Blender 4.0.

rmd6502, 2024-01-04:
cool, worked

closed 2024-01-04 by rmd6502