30350n/pcb2blender #43: ImportError: cannot import name 'add_dependencies' from 'pcb2blender_importer.blender_addon_utils' (unknown location

CWE-119, 2024-01-08:
```
bpy.data.window_managers["WinMan"].addon_search = ""
bpy.context.space_data.system_folders_active = 3
Modules Installed (pcb2blender_importer) from 'F:\\blenderPlugins\\pcb2blender_importer_2-8.zip' into 'C:\\Users\\User\\AppData\\Roaming\\Blender Foundation\\Blender\\4.0\\scripts\\addons'
Traceback (most recent call last):
  File "G:\Blender\4.0\scripts\modules\addon_utils.py", line 364, in enable
    mod = importlib.import_module(module_name)
  File "G:\Blender\4.0\python\lib\importlib\__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "C:\Users\User\AppData\Roaming\Blender Foundation\Blender\4.0\scripts\addons\pcb2blender_importer\__init__.py", line 16, in <module>
    from .blender_addon_utils import add_dependencies, register_modules_factory
ImportError: cannot import name 'add_dependencies' from 'pcb2blender_importer.blender_addon_utils' (unknown location)

```

30350n, 2024-01-08:
That is weird. Can you confirm `C:\\Users\\User\\AppData\\Roaming\\Blender Foundation\\Blender\\4.0\\scripts\\addons\\pcb2blender_importer\\blender_addon_utils\\__init__.py` exists/make a screenshot of it's contents? (Just open it in any text editor).

CWE-119, 2024-01-08:
the addons folder is empty, there is no folder no nth

CWE-119, 2024-01-08:
[image]
[image]

CWE-119, 2024-01-08:
i fixed it

30350n, 2024-01-08:
Oh nice! How?

CWE-119, 2024-01-09:
idk why the addons file were empty 
i checked the addons file from the source code , there were files 
so i redownloaded and reinstalled , done

30350n, 2024-01-09:
Hm okay, odd. Maybe they got flagged by your antivirus software or something. Glad it's resolved now.

closed 2024-01-09 by 30350n