30350n/pcb2blender #44: Exception on python Action plugin code

CWE-119, 2024-01-09:
it worked for once when i checked it on test file 
after that it never worked 
idk why
i reinstalled everything , all the plugins , it didnt worked , not even with the test file i used in the first test 
[image]

CWE-119, 2024-01-09:
[image]
[image]

CWE-119, 2024-01-09:
kicad v 7.0

30350n, 2024-01-09:
Try manually deleting the tempdir, that should fix it for now.
[image]

CWE-119, 2024-01-09:
dyamm thanks <3

closed 2024-01-09 by 30350n