30350n/pcb2blender #45: Eevie rendering

zevv, 2024-01-17:
Well, this is one *excellent* plugin, this saves me so much work and time, and the results are superior to what I usually care to make by hand. Thank you for making this?

I have two questions though:

- Renders look superior using cycles, but eevee only shows a grayscale model - is this intended behavior?
- For some reason my blender is having pretty tough time handing the imported PCB's, although I'm running a pretty beefy machine (Ryzen7 24 cores + AMD 6600 GPU). Is there a reason why the models are giving my computer a hard time?

Thanks again!

zevv, 2024-01-17:
Addendum: something else funny is happening: after importing a model, I can not seem to quit blender properly: I quit from the main menu, then the whole app freezes

```
Blender 4.0.1
	build date: 2023-11-17
	build time: 00:31:26
	build commit date: 2023-11-16
	build commit time: 16:40
	build hash: d0dd92834a08
	build platform: Linux
	build type: release
	build c flags:  -Wall -Werror=implicit-function-declaration -Wstrict-prototypes -Werror=return-type -Werror=vla -Wmissing-prototypes -Wno-char-subscripts -Wno-unknown-pragmas -Wpointer-arith -Wunused-parameter -Wwrite-strings -Wlogical-op -Wundef -Winit-self -Wmissing-include-dirs -Wno-div-by-zero -Wtype-limits -Wformat-signedness -Wrestrict -Wno-stringop-overread -Wno-stringop-overflow -Wnonnull -Wabsolute-value -Wuninitialized -Wredundant-decls -Wshadow -Wimplicit-fallthrough=5 -Wno-error=unused-but-set-variable  -std=gnu11 -msse -pipe -fPIC -funsigned-char -fno-strict-aliasing -ffp-contract=off -msse2  
	build c++ flags:  -Wuninitialized -Wredundant-decls -Wall -Wno-invalid-offsetof -Wno-sign-compare -Wlogical-op -Winit-self -Wmissing-include-dirs -Wno-div-by-zero -Wtype-limits -Werror=return-type -Wno-char-subscripts -Wno-unknown-pragmas -Wpointer-arith -Wunused-parameter -Wwrite-strings -Wundef -Wcomma-subscript -Wformat-signedness -Wrestrict -Wno-suggest-override -Wuninitialized -Wno-stringop-overread -Wno-stringop-overflow -Wimplicit-fallthrough=5 -Wundef -Wmissing-declarations  -msse -pipe -fPIC -funsigned-char -fno-strict-aliasing -ffp-contract=off -msse2  
	build link flags:  -Wl,--version-script='/home/blender/git/blender-v400/blender.git/source/creator/symbols_unix.map'
	build system: CMake

```

30350n, 2024-01-17:
Reason for both of these things are pretty much that the shaders aren't really optimized well. I want to rework them, but I didn't really find time for that yet. (See #24).

zevv, 2024-01-17:
Ok, clear answer, will close this issue then. Thanks!

closed 2024-01-17 by zevv

30350n, 2024-01-17:
> Addendum: something else funny is happening: after importing a model, I can not seem to quit blender properly: I quit from the main menu, then the whole app freezes
>
> ```
> Blender 4.0.1
> 	build date: [...]

No clue really about that one tbh, I'm doing some "*weird*" stuff with the materials, could be related to that.