30350n/pcb2blender #47: Error installing addon on blender

aeonSolutions, 2024-02-27:
Hi there @30350n 

I tried installing the addon on blender 4.0 (windows 11) , and spite the log showing  a successful installation in

Modules Installed () from 'C:\\Users\\Aeon\\Desktop\\pcb2blender_importer.zip' into 'C:\\Users\\Aeon\\AppData\\Roaming\\Blender Foundation\\Blender\\4.0\\scripts\\addons'"

the addon does not show anywhere for me to enable in the preferences dialog box .

What am i doing wrong ? 
thank you

30350n, 2024-02-27:
You probably didn't install the required dependencies in the Preferences window. Under Preferences -> Addons (search for pcb2blender).

aeonSolutions, 2024-02-27:
I did that however It shows nothing.

aeonSolutions, 2024-02-27:
Here's a snapshot of my screen 
[image]

30350n, 2024-02-27:
Where did you download the zip from? You have to download from the latest release.

aeonSolutions, 2024-02-27:
you are correct. I downloaded from the latest release.

30350n, 2024-02-27:
Did you rename the zip file then? Or unpack it and pack it again? The correct file from the latest release is named `pcb2blender_importer_2-9.zip`. You have to install that file in Blender.

aeonSolutions, 2024-02-28:
Solved. It turs out i was trying to install the exporter file . 
thank you

closed 2024-02-28 by aeonSolutions

30350n, 2024-02-28:
Oh yeah, that also doesn't work obviously 😅