30350n/pcb2blender #50: various issues with the generated files for oddish project

alinelena, 2024-03-10:
the plugin in general work great, better than the native export

few issues i noticed

- [ ] for smd footprints with tht pads in the smd pads i get joints, [1]
- [ ] missing pads for smd footprint, this is a sk6812, [2]
- [ ] missing elements in renders, eg screen and usc type c [3]
- [ ] would be nice to have an option to control the rendering scale, as export wrml has or have a note to say at what scale is exported.

this is the full kicad render
[image]

[1] [image]
[2] [image]
[3] [image]

the full repo is of the project is https://gitlab.com/m-lego/m65

exported file: 
[a.zip](https://github.com/30350n/pcb2blender/files/14551903/a.zip)

30350n, 2024-03-10:
Thanks for checking out the project!

[1] is due to invalid footprint setup most likely, the solder joints get created based per Pad, based on it's type (you probably got it set to THT).

[2] looks like one of these pad types: https://github.com/30350n/pcb2blender/blob/bfcc0f74eea43cc32263c28a81d7d3b4d2404c59/pcb2blender_importer/importer.py#L540-L542
which are not supported for generating solder joints automatically (you should get that warning in the console).

[3] atleast for that subboard on the right, it seems like KiCads WRL exporter only exports that part. So there's not much I can do about that I think.

> would be nice to have an option to control the rendering scale, as export wrml has or have a note to say at what scale is exported.

They are exported at the **correct** scale. They are just really small because by default Blender is zoomed out somewhere in the meters range. If you want to scales things up (which I'd generally not recommend) you can just select everything [Ctrl + A] and scale it up [S 100] (for example).

alinelena, 2024-03-11:
> [1] is due to invalid footprint setup most likely, the solder joints get created based per Pad, based on it's type (you probably got it set to THT).
yes, the footprint has two types of pads, smd and tht, i suggest the joints shall be drawn on the type of the component rather than purely on pad time.

> [2] looks like one of these pad types:
> 
> https://github.com/30350n/pcb2blender/blob/bfcc0f74eea43cc32263c28a81d7d3b4d2404c59/pcb2blender_importer/importer.py#L540-L542
> 
> 
> which are not supported for generating solder joints automatically (you should get that warning in the console).
yes indeed, it is a pitty they are not supported in diy boards having different pads is important.

> [3] atleast for that subboard on the right, it seems like KiCads WRL exporter only exports that part. So there's not much I can do about that I think.
I will double check them but I am sure export wrml exports more of them.

> They are exported at the **correct** scale. They are just really small because by default Blender is zoomed out somewhere in the meters range. If you want to scales things up (which I'd generally [...]
I am not saying are not exported at correct scale, would be nice to know the scale, printed as info. this is useful if you combine with other models non kicad.

30350n, 2024-03-11:
> yes, the footprint has two types of pads, smd and tht, i suggest the joints shall be drawn on the type of the component rather than purely on pad type.

That doesn't make any sense. If your footprints pads are set to the wrong type, I'd suggest you to fix that.

> yes indeed, it is a pitty they are not supported in diy boards having different pads is important.

Not sure what that's supposed to mean exactly, but you can always just add the few missing solder joints if you really care. (Realisticly, nobody is probably going to notice if you don't).

> I am not saying are not exported at correct scale, would be nice to know the scale, printed as info. this is useful if you combine with other models non kicad.

Once again, not sure what that's supposed to mean. Your board will be as big ... as your board is supposed to be. If your board is 20cm long, it'll be 20cm long in Blender. (Not sure how "printing" that would help).

alinelena, 2024-03-11:
> > yes, the footprint has two types of pads, smd and tht, i suggest the joints shall be drawn on the type of the component rather than purely on pad type.
> 
> That doesn't make any sense. If your footprints pads are set to the wrong type, I'd suggest you to fix that.
you can think of them as via in pad perfectly fine. help a beginner/diy to route things. 

> 
> > yes indeed, it is a pitty they are not supported in diy boards having different pads is important.
> 
> Not sure what that's supposed to mean exactly, but you can always just add the few missing solder joints if you really care. (Realisticly, nobody is probably going to notice if you don't).
I will look into it. 

> Once again, not sure what that's supposed to mean. Your board will be as big ... as your board is supposed to be. If your board is 20cm long, it'll be 20cm long in Blender. (Not sure how "printing" [...]
in export wrml there is this:
[image]

i suspect similarly your plugin makes one of these choices... just wanted to have it printed. 

the missing models seem to be to the fact that the plugin in the case of wrl models does not copy the models that are referenced by the models themselves. 
once I unzipped the pcb3d and added in the missing models in the components all seems to be ok
[image]

30350n, 2024-03-11:
> you can think of them as via in pad perfectly fine. help a beginner/diy to route things.

I don't understand what you are trying to say ...

> i suspect similarly your plugin makes one of these choices... just wanted to have it printed.

It doesn't really matter what option is chosen there, because things will be rescaled on import anyways, so that the scale is **correct** in Blender. How would printing that information be any help to you?

> the missing models seem to be to the fact that the plugin in the case of wrl models does not copy the models that are referenced by the models themselves.
once I unzipped the pcb3d and added in the missing models in the components all seems to be ok

Probably an issue with KiCads WRL exporter then. Will see if I can reproduce it.

closed 2025-05-24 by 30350n