30350n/pcb2blender #51: undefined shaders in blender 4.0

alinelena, 2024-03-12:
using blender 4.0 i get in shadding these

[image]

30350n, 2024-03-12:
Looks like you didn't properly install the addon.

alinelena, 2024-03-12:
what do you mean? followed the steps from the instructions... is there a way to check, the import function works correctly.

30350n, 2024-03-12:
Can you send a screenshot of the addon in the Preferences Window? It should look like this:

[image]

alinelena, 2024-03-12:
mine like this 
[image]

30350n, 2024-03-12:
Hmm that looks fine, did you get any errors during import? (Try to start Blender from a Terminal and check the output).

alinelena, 2024-03-12:
nope I am checking something... that was by saving the blender file that I have imported...

30350n, 2024-03-12:
Yeah, if you switch versions you might just have to reimport.

30350n, 2024-03-12:
> what happens the undefined happens when I move it on a machine with blender but without the plugin installed.

Yeah, that's expected.