30350n/pcb2blender #52: Solid black board when rendered on SheepIt Render Farm using Cycles

VitroidFPV, 2024-03-17:
As my PC is fairly slow I prefer to use SheepIt when possible. But it doesn't render properly if I use the `Enhance Materials` toggle when importing a board.
[image]

I tried packing all resources into the blend file, to no effect. Maybe I'm doing something wrong on my end, I'd appreciate any help regardless. It renders just fine locally, although slowly. This is still an amazing tool to get proper renders ❤️

30350n, 2024-03-23:
Sorry for the late answer ...
This is more or less a problem with the current material setup, which has to be reworked at some point (see #24). I use custom material nodes, which are basically just node groups with extra steps. For each custom node, there'll be a node group which is automatically generated, so in theory you should be able to use those. Switching all the material nodes for node groups manually would probably take a while though, ideally this would be automated.

Rootthecause, 2024-05-05:
I've encountered the same issue. Just saving the blend file and reopening it causes black textures. This makes it impossible to work on, share or backup the file. I'm using Blender 4.1 on MacOS.

30350n, 2024-05-05:
> I've encountered the same issue. Just saving the blend file and reopening it causes black textures. This makes it impossible to work on, share or backup the file. I'm using Blender 4.1 on MacOS.

That's a different issue probably. I'm guessing you are using Blender 4.1.**0**? That version has a bug which stops it from loading files using this addon. ([See the release notes](https://github.com/30350n/pcb2blender/releases/tag/v2.11-k8.0-b4.1)).

Rootthecause, 2024-05-05:
OH. That's it! I've missed this note 😓 
Thanks for the super fast reply! 
You've saved me more than enough time already! Thanks a lot!
Btw. the generated solder mesh looks incredible!

Snevzor, 2024-10-08:
> That's a different issue probably. I'm guessing you are using Blender 4.1.**0**? That version has a bug which stops it from loading files using this addon. ([See the release [...]

Hi, I'm having this issue with 4.2.2. Re-opening a saved file and all is black.

Snevzor, 2024-10-10:
My apologies for the late response. I had to test this multiple times to be able to provide useful information.
When Blender crashes after installing the extensions (for whatever reason), then the installation seems to be not 100% complete when starting Blender again. Uninstalling and re-installing fixes the render, but not the viewport. Restarting Blender after installing the extension seems to avoid this issue.
I presume this is a Blender issue and not a pcb2blender issue.

30350n, 2024-10-11:
> I presume this is a Blender issue and not a pcb2blender issue.

Yeah, mostly probably.

erikhole, 2024-10-21:
I'm also encountering this issue. It also sometimes turns the soldermask white

Rootthecause, 2024-10-21:
Intel Mac or M Series?
I'm using an an M1 Max and have no issues with Blender 4.2 LTS or 4.2.3 LTS with the latest PCB2Blender Plugins.

zshzebra, 2025-03-03:
I have this exact issue with a self hosted render farm.

Is there any (even hacky) solution to this?

[image]

30350n, 2025-03-03:
Installing the extension on the host used to render should fix it. If you are hosting it yourself, you should be able to do that I guess.

zshzebra, 2025-03-03:
> Installing the extension on the host used to render should fix it. If you are hosting it yourself, you should be able to do that I guess.

Thank you!