30350n/pcb2blender #54: Pcb2blender ersion 2.9 crash on Blender 4.1

eriklundh, 2024-03-29:
I had to downgrade to Blender 4.0.2 to get pcb2blender to work.
pcb2blender 2.9 has som issues in the install process on Blender 4.1,

30350n, 2024-03-29:
Blender 4.1 is not supported yet.

eriklundh, 2024-03-30:
No disrespect, fantastic tool.
Just wanted to report.

30350n, 2024-03-30:
Thanks, all good no worries! ^^

JosephDemarest, 2024-04-07:
> Blender 4.1 is not supported yet.

is there any update on a potential release time? My system runs like shit on blender 4.0, but on 4.1 it's great. and your tool is fantastic!

Juleinn, 2024-04-13:
Hello ! 
The issue seems to be with the way auto smooth is handled in blender 4.1 ( essentially the option has now been changed to shade smooth by angle, which marks edges as sharp)
```python
obj.data.use_auto_smooth = True
``` 
is no longer available (importer.py:1029)
```python
bpy.ops.object.shade_smooth_by_angle()
``` 
should work, but I don't know if that is what you want to do in this context.
Would you like me to submit a PR for this ?

Best regards,

30350n, 2024-04-14:
The auto smooth issue should be fixed now, sadly that [wasn't very straight forward though](https://github.com/30350n/pcb2blender/commit/9e4a0c64248650bfe2233728da461e0e7790486d) ...

My mediocre test setup is indicating that there's an issue with file saving/loading though, so there won't be a release until that's resolved too.

Juleinn, 2024-04-14:
Great, thanks you ! :)

30350n, 2024-04-15:
> My mediocre test setup is indicating that there's an issue with file saving/loading though, so there won't be a release until that's resolved too.

This is due to a bug in Blender 4.1.0 so that version won't be supported. A version that supports 4.1.1+ [is available now](https://github.com/30350n/pcb2blender/releases/tag/v2.10-k8.0-b4.1).

closed 2024-04-15 by 30350n