30350n/pcb2blender #56: OSError: load: /usr/bin/4.0/datafiles/assets/geometry_nodes/smooth_by_angle.blend failed to open blend file

alinelena, 2024-04-16:
updated to the last version of the plugin on kicad 8 and blender 4.0.1(linux) I get the above error.

full trace is 
```bash
Traceback (most recent call last):
  File "/home/drFaustroll/.config/blender/4.0/scripts/addons/pcb2blender_importer/importer.py", line 1047, in execute
    add_smooth_by_angle_modifier(context, obj)
  File "/home/drFaustroll/.config/blender/4.0/scripts/addons/pcb2blender_importer/importer.py", line 1097, in add_smooth_by_angle_modifier
    with bpy.data.libraries.load(SMOOTH_BY_ANGLE_ASSET_PATH) as (data_from, data_to):
OSError: load: /usr/bin/4.0/datafiles/assets/geometry_nodes/smooth_by_angle.blend failed to open blend file
Error: Python: Traceback (most recent call last):
  File "/home/drFaustroll/.config/blender/4.0/scripts/addons/pcb2blender_importer/importer.py", line 1047, in execute
    add_smooth_by_angle_modifier(context, obj)
  File "/home/drFaustroll/.config/blender/4.0/scripts/addons/pcb2blender_importer/importer.py", line 1097, in add_smooth_by_angle_modifier
    with bpy.data.libraries.load(SMOOTH_BY_ANGLE_ASSET_PATH) as (data_from, data_to):
OSError: load: /usr/bin/4.0/datafiles/assets/geometry_nodes/smooth_by_angle.blend failed to open blend file
Location: /usr/share/blender/4.0/scripts/modules/bpy/ops.py:109
Traceback (most recent call last):
  File "/home/drFaustroll/.config/blender/4.0/scripts/addons/pcb2blender_importer/importer.py", line 205, in execute
    if (pcb := self.import_pcb3d(context, filepath)) == {"CANCELLED"}:
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/drFaustroll/.config/blender/4.0/scripts/addons/pcb2blender_importer/importer.py", line 305, in import_pcb3d
    bpy.ops.pcb2blender.import_x3d(
  File "/usr/share/blender/4.0/scripts/modules/bpy/ops.py", line 109, in __call__
    ret = _op_call(self.idname_py(), kw)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
RuntimeError: Error: Python: Traceback (most recent call last):
  File "/home/drFaustroll/.config/blender/4.0/scripts/addons/pcb2blender_importer/importer.py", line 1047, in execute
    add_smooth_by_angle_modifier(context, obj)
  File "/home/drFaustroll/.config/blender/4.0/scripts/addons/pcb2blender_importer/importer.py", line 1097, in add_smooth_by_angle_modifier
    with bpy.data.libraries.load(SMOOTH_BY_ANGLE_ASSET_PATH) as (data_from, data_to):
OSError: load: /usr/bin/4.0/datafiles/assets/geometry_nodes/smooth_by_angle.blend failed to open blend file
Location: /usr/share/blender/4.0/scripts/modules/bpy/ops.py:109

```

30350n, 2024-04-16:
This got fixed in [`5bfdc01`](https://github.com/30350n/pcb2blender/commit/5bfdc019f8f36827fd92f85ebc3799435be3a2f9) a few hours ago (see #55). There's another bug in the remeshing code somewhere that I need to track down, will make a new release once that's fixed aswell.

30350n, 2024-04-18:
Can you confirm that this is fixed in [v2.11](https://github.com/30350n/pcb2blender/releases/tag/v2.11-k8.0-b4.1)?

alinelena, 2024-04-19:
not really, the path seems to be correct but still tries to load something that does not exist
```
Traceback (most recent call last):
  File "/home/drFaustroll/.config/blender/4.0/scripts/addons/pcb2blender_importer/importer.py", line 1083, in execute
    add_smooth_by_angle_modifier(obj)
  File "/home/drFaustroll/.config/blender/4.0/scripts/addons/pcb2blender_importer/importer.py", line 1139, in add_smooth_by_angle_modifier
    with bpy.data.libraries.load(SMOOTH_BY_ANGLE_ASSET_PATH) as (data_from, data_to):
OSError: load: /usr/share/blender/4.0/datafiles/assets/geometry_nodes/smooth_by_angle.blend failed to open blend file
Error: Python: Traceback (most recent call last):
  File "/home/drFaustroll/.config/blender/4.0/scripts/addons/pcb2blender_importer/importer.py", line 1083, in execute
    add_smooth_by_angle_modifier(obj)
  File "/home/drFaustroll/.config/blender/4.0/scripts/addons/pcb2blender_importer/importer.py", line 1139, in add_smooth_by_angle_modifier
    with bpy.data.libraries.load(SMOOTH_BY_ANGLE_ASSET_PATH) as (data_from, data_to):
OSError: load: /usr/share/blender/4.0/datafiles/assets/geometry_nodes/smooth_by_angle.blend failed to open blend file
Location: /usr/share/blender/4.0/scripts/modules/bpy/ops.py:109
Traceback (most recent call last):
  File "/home/drFaustroll/.config/blender/4.0/scripts/addons/pcb2blender_importer/importer.py", line 205, in execute
    if (pcb := self.import_pcb3d(context, filepath)) == {"CANCELLED"}:
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/drFaustroll/.config/blender/4.0/scripts/addons/pcb2blender_importer/importer.py", line 305, in import_pcb3d
    bpy.ops.pcb2blender.import_x3d(
  File "/usr/share/blender/4.0/scripts/modules/bpy/ops.py", line 109, in __call__
    ret = _op_call(self.idname_py(), kw)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
RuntimeError: Error: Python: Traceback (most recent call last):
  File "/home/drFaustroll/.config/blender/4.0/scripts/addons/pcb2blender_importer/importer.py", line 1083, in execute
    add_smooth_by_angle_modifier(obj)
  File "/home/drFaustroll/.config/blender/4.0/scripts/addons/pcb2blender_importer/importer.py", line 1139, in add_smooth_by_angle_modifier
    with bpy.data.libraries.load(SMOOTH_BY_ANGLE_ASSET_PATH) as (data_from, data_to):
OSError: load: /usr/share/blender/4.0/datafiles/assets/geometry_nodes/smooth_by_angle.blend failed to open blend file
Location: /usr/share/blender/4.0/scripts/modules/bpy/ops.py:109
```

is there some other plugin that needs installed, or maybe that is on newer blender versions?

alinelena, 2024-04-19:
copying the file by hand, after getting it from blender seems to have fixed it

30350n, 2024-04-19:
You are using Blender 4.0 which is not supported. Only 4.1.1+ is supported.

closed 2024-05-17 by 30350n