30350n/pcb2blender #57: Blender 4.1.1 installation OK but no option in Import menu

ddB0515, 2024-04-17:
Hello all

I have Blender 4.1.1 and downloaded and installed latest version 2.10 installed in Blender and everything is OK
[image]

But when I go under File-Import there is no PCB (.pcb3d) option
[image]

I have Python 3.10.12 installed and don't know what else to do (tried uninstall, reinstall) but same thing every time
Anyone have same issue or it's just me :(

30350n, 2024-04-17:
Can you start Blender from a terminal and check if any errors come up?

ddB0515, 2024-04-18:
Sorry for late response, here is log from commandline
`$ ./blender
Read prefs: "/home/dale/.config/blender/4.1/config/userpref.blend"
home_builder addon: Verbose is enabled
home_builder addon: Read in JSON settings from file
Traceback (most recent call last):
  File "/home/dale/apps/blender-4.1.1-linux-x64/4.1/scripts/modules/addon_utils.py", line 376, in enable
    mod = importlib.import_module(module_name)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/dale/apps/blender-4.1.1-linux-x64/4.1/python/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/dale/.config/blender/4.1/scripts/addons/pcb2blender_importer/__init__.py", line 26, in <module>
    register, unregister = register_modules_factory(modules)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/dale/.config/blender/4.1/scripts/addons/pcb2blender_importer/blender_addon_utils/register.py", line 37, in register_modules_factory
    _modules.append(importlib.import_module(f".{module}", package=caller_package))
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/dale/apps/blender-4.1.1-linux-x64/4.1/python/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/dale/.config/blender/4.1/scripts/addons/pcb2blender_importer/importer.py", line 9, in <module>
    from PIL import Image, ImageOps
  File "/home/dale/.config/blender/4.1/scripts/addons/home_builder_4/python_libs/PIL/Image.py", line 103, in <module>
    from . import _imaging as core
ImportError: cannot import name '_imaging' from 'PIL' (/home/dale/.config/blender/4.1/scripts/addons/home_builder_4/python_libs/PIL/__init__.py)
Warning: add-on missing 'bl_info', this can cause poor performance!: '/home/dale/.config/blender/4.1/scripts/addons/plugins/__init__.py'
Traceback (most recent call last):
  File "/home/dale/apps/blender-4.1.1-linux-x64/4.1/scripts/modules/addon_utils.py", line 376, in enable
    mod = importlib.import_module(module_name)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/dale/apps/blender-4.1.1-linux-x64/4.1/python/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/dale/.config/blender/4.1/scripts/addons/pcb2blender_importer/__init__.py", line 26, in <module>
    register, unregister = register_modules_factory(modules)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/dale/.config/blender/4.1/scripts/addons/pcb2blender_importer/blender_addon_utils/register.py", line 37, in register_modules_factory
    _modules.append(importlib.import_module(f".{module}", package=caller_package))
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/dale/apps/blender-4.1.1-linux-x64/4.1/python/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/dale/.config/blender/4.1/scripts/addons/pcb2blender_importer/importer.py", line 9, in <module>
    from PIL import Image, ImageOps
  File "/home/dale/.config/blender/4.1/scripts/addons/home_builder_4/python_libs/PIL/Image.py", line 103, in <module>
    from . import _imaging as core
ImportError: cannot import name '_imaging' from 'PIL' (/home/dale/.config/blender/4.1/scripts/addons/home_builder_4/python_libs/PIL/__init__.py)
`

30350n, 2024-04-18:
(Logs in text form are generally more useful because you can search/copy from them).

ddB0515, 2024-04-18:
Here is log version :)
`dale@machine:~/apps/blender-4.1.1-linux-x64$ ./blender
Read prefs: "/home/dale/.config/blender/4.1/config/userpref.blend"
home_builder addon: Verbose is enabled
home_builder addon: Read in JSON settings from file
Traceback (most recent call last):
  File "/home/dale/apps/blender-4.1.1-linux-x64/4.1/scripts/modules/addon_utils.py", line 376, in enable
    mod = importlib.import_module(module_name)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/dale/apps/blender-4.1.1-linux-x64/4.1/python/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/dale/.config/blender/4.1/scripts/addons/pcb2blender_importer/__init__.py", line 26, in <module>
    register, unregister = register_modules_factory(modules)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/dale/.config/blender/4.1/scripts/addons/pcb2blender_importer/blender_addon_utils/register.py", line 37, in register_modules_factory
    _modules.append(importlib.import_module(f".{module}", package=caller_package))
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/dale/apps/blender-4.1.1-linux-x64/4.1/python/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/dale/.config/blender/4.1/scripts/addons/pcb2blender_importer/importer.py", line 9, in <module>
    from PIL import Image, ImageOps
  File "/home/dale/.config/blender/4.1/scripts/addons/home_builder_4/python_libs/PIL/Image.py", line 103, in <module>
    from . import _imaging as core
ImportError: cannot import name '_imaging' from 'PIL' (/home/dale/.config/blender/4.1/scripts/addons/home_builder_4/python_libs/PIL/__init__.py)
Warning: add-on missing 'bl_info', this can cause poor performance!: '/home/dale/.config/blender/4.1/scripts/addons/plugins/__init__.py'
Traceback (most recent call last):
  File "/home/dale/apps/blender-4.1.1-linux-x64/4.1/scripts/modules/addon_utils.py", line 376, in enable
    mod = importlib.import_module(module_name)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/dale/apps/blender-4.1.1-linux-x64/4.1/python/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/dale/.config/blender/4.1/scripts/addons/pcb2blender_importer/__init__.py", line 26, in <module>
    register, unregister = register_modules_factory(modules)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/dale/.config/blender/4.1/scripts/addons/pcb2blender_importer/blender_addon_utils/register.py", line 37, in register_modules_factory
    _modules.append(importlib.import_module(f".{module}", package=caller_package))
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/dale/apps/blender-4.1.1-linux-x64/4.1/python/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/dale/.config/blender/4.1/scripts/addons/pcb2blender_importer/importer.py", line 9, in <module>
    from PIL import Image, ImageOps
  File "/home/dale/.config/blender/4.1/scripts/addons/home_builder_4/python_libs/PIL/Image.py", line 103, in <module>
    from . import _imaging as core
ImportError: cannot import name '_imaging' from 'PIL' (/home/dale/.config/blender/4.1/scripts/addons/home_builder_4/python_libs/PIL/__init__.py)
`

30350n, 2024-04-18:
Seems like the `home_builder_4` addon is causing the issue. From what I can tell the developer decided to directly include the required Python dependencies (which is a terrible idea). So the version of `PIL` that gets loaded is outdated.

30350n, 2024-04-18:
If you uninstall that addon, `pcb2blender` should work fine.

ddB0515, 2024-04-18:
Indeed that's an issue, removed it and works fine
Thanks for suggestion :)

closed 2024-04-18 by ddB0515