30350n/pcb2blender #58: What does this add?

marcdraco, 2024-04-28:
I know I might sound like an old farty here but I failed to get this to work until I downloaded your importer and after all the fooling around with text markers, what came out of Blender was nothing like what came out of PCB New.

But more than that, PCB New already has this function and while it's buggy, it works directly in Blender with the existing import/export, it's faster as a result and you just export the board directly.

I'm a developer myself so I'm cognisant of people just jumping in and complaining. I'm honesty confused why you did this rather than extend the existing exporter? Multi-stacks perhaps?

The only issue I had with the existing export in PCB New is it's necessary to join all the parts into one to to transforms because each object has its own origin. Honestly man, I'm not taking  pop, I'm just confused.

30350n, 2024-04-29:
### KiCad 3D Viewer:

[image]

### Blender import (without pcb2blender, VRML format):

[image]

### Blender import using pcb2blender:

[image]

(Both Blender screenshots are straight out of import, the only thing I added is a HDRI for lighting. The pcb2blender render took approximately twice as long to run, compared to the KiCad render (~3s vs ~6s on a 9 year old gpu), which for the increase in quality is more than fine).

> I downloaded your importer and after all the fooling around with text markers

If you are only exporting a single (non panelized) PCB, you can just skip that.

> I'm honesty confused why you did this rather than extend the existing exporter?

What "exporter"?

> The only issue I had with the existing export in PCB New is it's necessary to join all the parts into one to to transforms because each object has its own origin.

That's more of a feature than an issue, the main problem here is that Blender's VRML importer doesn't join each individual part (which might consist of multiple face sets) together. pcb2blender fixes this.

30350n, 2024-05-13:
Did that sufficiently answer your questions?

closed 2024-05-17 by 30350n

marcdraco, 2024-06-15:
Yeah, thanks it did. Sorry I don't get in much.

I did notice your shaders absolutely killing my older machines during compilation, but that's just Blender.

30350n, 2024-06-15:
All good, yeah they are pretty heavy. Optimizing them is on the TODO list (and has been for a while ...).

30350n, 2024-06-17:
If you are rendering on your CPU, there's your mistake. It will generally work, but it'll be very slow (and inefficient too) compared to using a dedicated GPU.

> even a "small" model

What matters here is detail not size.

marcdraco, 2024-06-17:
I meant the shader compilation, not the rendering. That's what brings the thing to its knees even just switching into the Evee preview starts all that messing around. Not your fault. just the way Blender works.

Funny thing about CPU vs. GPU. It's actually more efficient to use multiple CPUs with loads of memory than a dedicated graphics card for larger scenes using Cycles. I didn't believe it either but my 16 core Xeon machine (which is pretty ancient now) rendered in CPU mode almost as fast as the 1080TI (I've got a lot of tech just lying around), much of it it quite old. Smaller scenes are great on the GPU of course.

30350n, 2024-06-17:
> I meant the shader compilation, not the rendering. That's what brings the thing to its knees even just switching into the Evee preview starts all that messing around. Not your fault. just the way Blender works.

Partially my fault, because the custom shaders are really not optimized for eevee, but yeah.

> It's actually more efficient to use multiple CPUs with loads of memory than a dedicated graphics card for larger scenes using Cycles. I didn't believe it either but my 16 core Xeon machine (which is [...]

Huh, are you using CUDA or Optix with the 1080TI?

marcdraco, 2024-06-18:
CUDA for the Ti but that machine is so power hungry that it rarely gets switched on these days. The advice came from (Blender) Ton himself at Blendercon a few years back.

I thought he'd lost his chips to be honest, but the limited memory of GPUs (mine is 12Gb) was the issue I believe. I wear rather more hats than my poor little noodle can comfortably handle so I've got to manage with a limited skillset on each if that makes sense.