30350n/pcb2blender #59: Cannot install v2.7 in Blender 3.6

VanessaE, 2024-05-05:
Release v2.7 seems the last version that's supposed to work with Blender 3.6<sup>[*]</sup> as far as I can see, but I just can't seem to install this fully/properly.  

All of the following were done with a freshly extracted Blender build, from the official Linux `.xz` archive.

I downloaded your v2.7 release zip and installed it through the add-ons manager as one does with any other addon.  Nothing notable in the terminal, just one line saying the add-on was installed.

After installing, I was presented with this when I tried to enable it:

[image]

Using `pip` to install packages globally has been strongly discouraged for a long time now (a few years I guess).  In fact, the way I understand it, even `pip` itself shouldn't be installed globally.  One is supposed to use a virtual environment -- but Blender doesn't use a typical virtual environment at all (probably because it is its own environment to begin with).

Okay... well, it's unclear what would happen if I clicked the "install" button there, so I'll exit Blender and head back to the command line.

My system (Debian Bookworm) uses 3.11 normally, and Blender 3.6 ships with Python 3.10, so maybe if use that version directly, it'll just download and "install" `pip` locally to the Blender folder.  Assuming that, I downloaded the `get-pip.py` script from the main site, and ran it thus:

```bash
vanessa@rainbird:/data/Archives/Software/Linux/blender-related/blender-3.6.1-linux-x64$ cd 3.6/python/bin/
vanessa@rainbird:/data/Archives/Software/Linux/blender-related/blender-3.6.1-linux-x64/3.6/python/bin$ ./python3.10 /data/Archives/Software/Linux/get-pip.py 
Collecting pip
  Using cached pip-24.0-py3-none-any.whl.metadata (3.6 kB)
Collecting wheel
  Using cached wheel-0.43.0-py3-none-any.whl.metadata (2.2 kB)
Using cached pip-24.0-py3-none-any.whl (2.1 MB)
Using cached wheel-0.43.0-py3-none-any.whl (65 kB)
Installing collected packages: wheel, pip
  Attempting uninstall: pip
    Found existing installation: pip 23.0.1
    Uninstalling pip-23.0.1:
      Successfully uninstalled pip-23.0.1
Successfully installed pip-24.0 wheel-0.43.0
```

Okay, I suppose that output looks right.  As above, your add-on said I need `skia-python` and `pillow`, whatever those are.  Okay, let's use that newly-installed `pip`:

```bash
vanessa@rainbird:/data/Archives/Software/Linux/blender-related/blender-3.6.1-linux-x64/3.6/python/bin$ ./pip3.10 install skia-python pillow
Collecting skia-python
  Using cached skia_python-87.5-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (2.9 kB)
Collecting pillow
  Using cached pillow-10.3.0-cp310-cp310-manylinux_2_28_x86_64.whl.metadata (9.2 kB)
Requirement already satisfied: numpy in /home/data/Archives/Software/Linux/blender-related/blender-3.6.1-linux-x64/3.6/python/lib/python3.10/site-packages (from skia-python) (1.23.5)
Collecting pybind11>=2.6 (from skia-python)
  Using cached pybind11-2.12.0-py3-none-any.whl.metadata (9.5 kB)
Using cached skia_python-87.5-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (10.8 MB)
Using cached pillow-10.3.0-cp310-cp310-manylinux_2_28_x86_64.whl (4.5 MB)
Using cached pybind11-2.12.0-py3-none-any.whl (234 kB)
Installing collected packages: pybind11, pillow, skia-python
Successfully installed pillow-10.3.0 pybind11-2.12.0 skia-python-87.5
```

That seems right.  Let's run Blender:

```bash
vanessa@rainbird:/data/Archives/Software/Linux/blender-related/blender-3.6.1-linux-x64/3.6/python/bin$ cd ../../..
vanessa@rainbird:/data/Archives/Software/Linux/blender-related/blender-3.6.1-linux-x64$ ./blender
Read prefs: "/home/vanessa/.config/blender/3.6/config/userpref.blend"
Traceback (most recent call last):
  File "/home/data/Archives/Software/Linux/blender-related/blender-3.6.1-linux-x64/3.6/scripts/modules/addon_utils.py", line 333, in enable
    mod = __import__(module_name)
  File "/home/vanessa/.config/blender/3.6/scripts/addons/pcb2blender_importer/__init__.py", line 26, in <module>
    register, unregister = register_modules_factory(modules)
  File "/home/vanessa/.config/blender/3.6/scripts/addons/pcb2blender_importer/blender_addon_utils/register.py", line 35, in register_modules_factory
    _modules.append(importlib.import_module(f".{module}", package=caller_package))
  File "/home/data/Archives/Software/Linux/blender-related/blender-3.6.1-linux-x64/3.6/python/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "/home/vanessa/.config/blender/3.6/scripts/addons/pcb2blender_importer/importer.py", line 1075, in <module>
    del namespace["bl_rna"]
KeyError: 'bl_rna'
```
[... 57 lines cut ...]
Modules Installed (pcb2blender_importer) from '/data/Archives/Software/Linux/blender-related/add-ons/pcb2blender_importer_2-7.zip' into '/home/vanessa/.config/blender/3.6/scripts/addons'
Traceback (most recent call last):
  File "/home/data/Archives/Software/Linux/blender-related/blender-3.6.1-linux-x64/3.6/scripts/modules/addon_utils.py", line 333, in enable
    mod = __import__(module_name)
  File "/home/vanessa/.config/blender/3.6/scripts/addons/pcb2blender_importer/__init__.py", line 26, in <module>
    register, unregister = register_modules_factory(modules)
  File "/home/vanessa/.config/blender/3.6/scripts/addons/pcb2blender_importer/blender_addon_utils/register.py", line 35, in register_modules_factory
    _modules.append(importlib.import_module(f".{module}", package=caller_package))
  File "/home/data/Archives/Software/Linux/blender-related/blender-3.6.1-linux-x64/3.6/python/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "/home/vanessa/.config/blender/3.6/scripts/addons/pcb2blender_importer/importer.py", line 1075, in <module>
    del namespace["bl_rna"]
KeyError: 'bl_rna'
```

Failed again.

What did I miss?

Sorry if this comes off a little coarse.  I just spent the last few hours arguing with KiCAD -- a program I have never used (I last used Lepton/gEDA) -- to get the PCB in question into a useful state, an act that should have taken only a few minutes, and discovering a whole new level of frustration in the process.  That's all before I even came across this project.

-----
[*] I can't use Blender 4.x due to an incompatibility with one of my projects.

Related: #13 #14

30350n, 2024-05-13:
> Using pip to install packages globally has been strongly discouraged for a long time now (a few years I guess). In fact, the way I understand it, even pip itself shouldn't be installed globally. One [...]

Okay... well, it's unclear what would happen if I clicked the "install" button there (would it be global? will it need sudo? or would be it be local? if that's the case, where would the files go?), so I'll exit Blender and head back to the command line.

The modules will be installed into a `site-packages` folder **inside** the addon directory. The path to said directory will then be prepended to `sys.path` during addon initialization. That's the best I could come up with (as you figured, Blender doesn't really support venvs).

(Blender *should* (some distros (for whatever reason) deviate from this) also come with it's own Python installation. If your distro doesn't do this, that could cause problems).

Anyways.

### The actual error you are getting

... seems to be related to my override of Blender's integrated `X3D/VRML2` importer. Make sure it's present and enabled.

[image]

VanessaE, 2024-05-14:
> The modules will be installed into a `site-packages` folder **inside** the `pcb2blender_importer` addon directory. The path to said directory will then be prepended to `sys.path` during addon initialization. That's the best I could come up with (as you figured, Blender doesn't really support venvs).

Noted.

> (Blender _should_ (some distros (for whatever reason) deviate from this) also come with it's own Python installation. If your distro doesn't do this, that could cause problems).

Though there's probably a recent-ish version in Debian Bookworm's repositories, I actually use the zipped build distributed directly by the Blender team.  Folks call Debian "stale" for a reason. 😛

> ### The actual error you are getting
> ... seems to be related to my override of Blender's integrated `X3D/VRML2` importer. Make sure it's present and enabled.

Ah hah, enabling that made the importer plugin behave at least enough to try to use it.

However, when I follow the readme to specify the board's bounds in Kicad, then export it, I get a 2.7 MB file, so that seems legit:

[image]

If I try to import that file into Blender, it hangs for several seconds, then I get a ginormous wireframe of one of the components along with a really tiny board:

[image]

[image]

The board must be exporting with meter resolution but the importer obvious expects millimeters, as the size is off by exactly 1000x.  Unfortunately, without components on it, it's not directly usable:

[image]

And I get this after it finished:

```
Python: Traceback (most recent call last):
  File "/home/vanessa/.config/blender/3.6/scripts/addons/pcb2blender_importer/importer.py", line 205, in execute
    if (pcb := self.import_pcb3d(context, filepath)) == {"CANCELLED"}:
  File "/home/vanessa/.config/blender/3.6/scripts/addons/pcb2blender_importer/importer.py", line 345, in import_pcb3d
    bpy.ops.pcb2blender.import_x3d(
  File "/home/data/Archives/Software/Linux/blender-related/blender-3.6.1-linux-x64/3.6/scripts/modules/bpy/ops.py", line 113, in __call__
    ret = _op_call(self.idname_py(), None, kw)
RuntimeError: Error: Python: Traceback (most recent call last):
  File "/home/vanessa/.config/blender/3.6/scripts/addons/pcb2blender_importer/importer.py", line 1011, in execute
    result = import_x3d.load(context, self.filepath, global_matrix=matrix)
  File "/home/data/Archives/Software/Linux/blender-related/blender-3.6.1-linux-x64/3.6/scripts/addons/io_scene_x3d/import_x3d.py", line 3629, in load
    load_web3d(context, filepath,
  File "/home/data/Archives/Software/Linux/blender-related/blender-3.6.1-linux-x64/3.6/scripts/addons/io_scene_x3d/import_x3d.py", line 3528, in load_web3d
    importShape(bpycollection, node, ancestry, global_matrix)
  File "/home/data/Archives/Software/Linux/blender-related/blender-3.6.1-linux-x64/3.6/scripts/addons/io_scene_x3d/import_x3d.py", line 3142, in importShape
    bpydata = geom_fn(geom, ancestry)
  File "/home/data/Archives/Software/Linux/blender-related/blender-3.6.1-linux-x64/3.6/scripts/addons/io_scene_x3d/import_x3d.py", line 1894, in importMesh_IndexedFaceSet
    if len(points) >= 2 * len(index):  # Need to cull
TypeError: object of type 'NoneType' has no len()
Location: /home/data/Archives/Software/Linux/blender-related/blender-3.6.1-linux-x64/3.6/scripts/modules/bpy/ops.py:113
```

30350n, 2024-05-14:
> Ah hah, enabling that made the importer plugin behave at least enough to try to use it.

Great!

>However, when I follow the readme to specify the board's bounds in Kicad, then export it, I get a 2.7 MB file, so that seems legit:

Note: You don't actually have to specify bounds for single board designs (it *shouldn't* hurt but it actually sometimes does 😅 ).

> If I try to import that file into Blender, it hangs for several seconds, then I get a ginormous wireframe of one of the components along with a really tiny board:
>The board must be exporting with meter resolution but the importer apparently expects millimeters, as the size is off by exactly 1000x. Unfortunately, without components on it, it's not directly usable:

What you are seing here doesn't really matter (it's only intermediate), as the importer is crashing during operation.

### Regarding the actual error:

The X3D/VRML importer seems to fail importing when importing that model (the micro SD card slot or whatever that is). Either that model is broken/invalid or there's a bug in the X3D/VRML importer (probably both). There's not much I can do about that, you can possibly try editing/fixing that model or just remove it.

30350n, 2024-05-14:
Oh also, out of curiosity, was the X3D/VRML importer addon not enabled by default with your debian installation?

VanessaE, 2024-05-14:
> Note: You don't actually have to specify bounds for single board
> designs (it shouldn't hurt but it actually sometimes does 😅 ).

Nope, I tried that first, it could not detect the board without my marking the bounds with those text items.

> What you are seing here doesn't really matter (it's only intermediate),
> as the importer is crashing during operation.

That would explain the lack of models on the board.

> Either that model is broken/invalid or there's a bug in the X3D/VRML 
> importer (probably both). There's not much I can do about that

Well, I found a workaround already (days ago) that doesn't involve your plugins: export the board to STEP, then use Freecad to import it:

[image]

So the SD card slot isn't broken, though there was some odd warning about "non-unity scaling" of that model when I exported it from Kicad.  

Then I exported that to STL and brought it into Blender.  STL of course loses all the materials and groupings, but it was enough to at least have something accurate to model my project against.

I can probably combine that with this partially-imported board from your plugins to at least improve my model's general appearance (e.g. to get the silk, pads/vias/holes, etc).  I'll putter around with that more later on.

> Oh also, out of curiosity, was the X3D/VRML importer
> addon not enabled by default with your debian installation?

It was disabled by default.

30350n, 2024-05-15:
> Nope, I tried that first, it could not detect the board without my marking the bounds with those text items.

Yeah ... the exporter UI is/was kind of not ideal. It says it doesn't detect any boards, but it will still just default to exporting everything.

> Well, I found a workaround already (days ago) that doesn't involve your plugins: export the board to STEP, then use Freecad to import it.

Yeah that's one manual way to do it (getting the models into Blender is the easy part really). Achieving a result that's visually similar to what this addon provides though, will probably take quite a lot of additional time (that's why it exists 😅 ).

If you are interested, [I did a quick comparison](https://github.com/30350n/pcb2blender/issues/58#issuecomment-2082189710) of the KiCad 3D Viewer, a vanilla VRML Blender import with default materials and pcb2blender a couple of weeks ago.

30350n, 2024-05-15:
> I can probably combine that with this partially-imported board from your plugins to at least improve my model's general appearance (e.g. to get the silk, pads/vias/holes, etc). I'll putter around with that more later on.

Also note: It's not really partially-imported either, as most of the magic happens **after** importing all the basic models. You'll really get a much better result if you disable tthe model causing the issue before exporting.

VanessaE, 2024-05-15:
> Yeah that's one manual way to do it (getting the models into Blender is the
> easy part really). Achieving a result that's visually similar to what this addon
> provides though, will probably take quite a lot of additional time (that's why it
> exists 😅 ).

Yeah I kinda got frustrated by the end, but this is what I eventually ended up with originally:

[image]

Combined with your importer's efforts yields this:

[image]

I can live with that. 🙂

closed 2024-05-17 by 30350n