30350n/pcb2blender #61: Cant export or inport

BretterBote, 2024-05-26:
I am getting an Error in KiCAD and Blender. In Blender when i activate the plugin, nothing shows up under File > Import. I started Blender as an Admin and got an Error. Same happened when i tried to export the pcb3d from KiCAD. Couldn´t copy the text so i took a screenshot. 

[image]
[image]

30350n, 2024-05-26:
The second error is due to insufficient permissions to open the path you've selected for writing. Either you're trying to export to some system directory or you've got messed up permissions on your system.

30350n, 2024-05-26:
First one could either be a collision with another addon or something messed up while installing the dependencies, which makes it so that PIL is detected as being installed, but the installation is corrupted.
Either way you shouldn't (have to) start Blender with elevated permissions.

30350n, 2024-09-21:
Closing this, as it should be resolved by installing the newest [v2.13](https://github.com/30350n/pcb2blender/releases/tag/v2.13-k8.0-b4.2) version which uses the new [Blender Extension Format](https://extensions.blender.org/about/)!

closed 2024-09-21 by 30350n