30350n/pcb2blender #62: Plugin fails to load

blackbit42, 2024-05-29:
Hi

I am using blender 3.4.1+dfsg-2+b1 on Debian stable.
The plugin installed alright.
[image]
The `File > Import` menu doesn't have a 'PCB' entry. I believe that is due to:
```
$ blender
Read prefs: /home/blackbit/.config/blender/3.4/config/userpref.blend
Traceback (most recent call last):
  File "/usr/share/blender/scripts/modules/addon_utils.py", line 333, in enable
    mod = __import__(module_name)
          ^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/blackbit/.config/blender/3.4/scripts/addons/pcb2blender_importer/__init__.py", line 26, in <module>
    register, unregister = register_modules_factory(modules)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/blackbit/.config/blender/3.4/scripts/addons/pcb2blender_importer/blender_addon_utils/register.py", line 37, in register_modules_factory
    _modules.append(importlib.import_module(f".{module}", package=caller_package))
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/blackbit/.config/blender/3.4/scripts/addons/pcb2blender_importer/importer.py", line 1131, in <module>
    get_internal_asset_path() / "geometry_nodes" / "smooth_by_angle.blend"
    ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/blackbit/.config/blender/3.4/scripts/addons/pcb2blender_importer/importer.py", line 1128, in get_internal_asset_path
    assert False
AssertionError
```

How can I determine why this assertion fails?

30350n, 2024-05-29:
The newest version of the plugin only supports Blender 4.1.1+.

closed 2024-05-29 by 30350n

30350n, 2024-05-30:
Blender itself should already do that. (The warning might not be super obvious I guess).