30350n/pcb2blender #64: Dark stripes/shadows on PCB

Fivefold, 2024-05-31:
I get some weird dark stripes on the exported/imported pcbs. They don't seem to be lighting-related, as they persist for different lighting conditions and even when removing all components but the PCB itself.

It could be a blender issue since I'm a total beginner in blender, but even if I start a new project and import the pcb I can see the stripes. They also always stay in the same locations.

Not sure how to even tackle this. I can send you the blender files privately if it helps to track down the issue.

<details><summary>Show images</summary>

[image]

[image]

[image]

</details>

30350n, 2024-06-01:
Probably an issue with the Boolean modifier again ...

If you are only using a single PCB, you don't have to specify it's bounds in KiCad, which should fix this.

For investigating the issue though: It'd be nice if you could share the `.pcb3d` file (if you want to do so privately, you can find my discord, e-mail and mastodon on [my personal website](https://30350n.de/)).

Fivefold, 2024-06-01:
I _am_ using single PCBs and I haven't specified it's bounds (in case you mean with the PCB3D_<TL/BR>_<boardname> text). Due to your suggestion I tried specifying the bounds and I don't see any stripes there! So there's a workaround.

I've sent you a request on discord to send you the .pcb3d file.

30350n, 2024-06-03:
This is fixed in [v2.12](https://github.com/30350n/pcb2blender/releases/tag/v2.12-k8.0-b4.1)!

closed 2024-06-03 by 30350n