30350n/pcb2blender #65: skia-python isn't detected albeit present

blackbit42, 2024-06-23:
Since it was advised in https://github.com/30350n/pcb2blender/issues/62 that my blender was too old, I now created an environment in which I can run a new enough blender (4.1.1+dfsg-2+b1).

Unfortunately that setup is not as straight forward as I would like it to be and I ran into an issue.

Setup:
* Debian unstable/experimental chroot - This allows me to install a new enough blender without messing up my main Debian installation. Blender works as expected in the chroot.
* Python dependencies are installed in the chroot via `venv` and `pip` because not all dependencies are available via debian packages. (skia-python) is missing.

I've installed all required dependencies
* requests (which is an undocumented dependency by the way)
* numpy
* pillow
* skia-python

The first three entries in the list above are detected as expected, but skia-python isn't and I struggle to understand why.

I run blender like so:
```PYTHONPATH=$HOME/.venv/blender_deps/lib/python3.12/site-packages/ blender --python-use-system-env```

the following works just fine:
```PYTHONPATH=$HOME/.venv/blender_deps/lib/python3.12/site-packages/:$HOME/.config/blender/4.1/scripts/addons/pcb2blender_importer/site-packages/ python3 -c 'from skia import SVGDOM, Color4f, Stream, Surface'```

`skia-python` is in the same `site-packages` directory as all other dependencies. I am not sure why it isn't detected.

30350n, 2024-06-26:
What's the specific error you are getting?

Any specific reason why you aren't using the automatic dependency installation mechanism this addon provides?

> requests (which is an undocumented dependency by the way)

If packaged correctly, Blender should come with it's own python installation which already comes with `requests` and `numpy` installed.

Also note: The correct python version for Blender 4.1 is [3.11.7](https://github.com/blender/blender/blob/blender-v4.1-release/build_files/build_environment/cmake/versions.cmake#L233).

blackbit42, 2024-06-26:
> What's the specific error you are getting?

It's difficult to respond to that question in a straight-forward fashion.
I worked through multiple issues with undetected / missing dependencies that I was able to address myself.
Eventually I ended up with the situation described above.
The addon dialog of blender now reports this:
[image]
There are no python backtraces or something like that during startup of blender.
Debian stopped to support `pip` as such a while ago.
I have pip installed, but depressing the 'Install skia-python' button doesn't yield the intended effect, unsurprisingly.
What is surprising though is that blender believes skia-python isn't there. In fact it is, as outlines above.

> If packaged correctly, Blender should come with it's own python installation

This doesn't seem to be the case on Debian.
The blender package doesn't depend on either `requests` or `numpy`.
Can you provide a reference to the blender requirement? I am happy to open a Debian bug to get this fixed.

> Also note: The correct python version for Blender 4.1 is 3.11.7.

The python version on my system is `3.11.2`.

30350n, 2024-06-26:
> What is surprising though is that blender believes skia-python isn't there. In fact it is, as outlines above.

That UI in the preferences is not native, but comes from my addon (from the [`blender_addon_utils`](https://github.com/30350n/blender_addon_utils/) submodule to be exact. To check if skia is installed it'd checking if [`importlib_util.find_spec("skia")`](https://github.com/30350n/blender_addon_utils/blob/1dbd149ff369eda297beb60da2780d15ad6254d0/register.py#L81) returns something. Not sure why that doesn't work in your case.

30350n, 2024-06-26:
>> If packaged correctly, Blender should come with it's own python installation
>
> This doesn't seem to be the case on Debian.

Are you sure about that?
If you launch Blender normally (without `--python-use-system-env`) and open a python console in it, what does `sys.executable` evaluate to?

blackbit42, 2024-06-26:
> The python version on my system is 3.11.2.

That was an incorrect statement.
My unstable chroot has python 3.12.4.

> To check if skia is installed it'd checking if importlib_util.find_spec("skia") returns something. Not sure why that doesn't work in your case.

```
$ PYTHONPATH=$HOME/.venv/blender_deps/lib/python3.12/site-packages/:$HOME/.config/blender/4.1/scripts/addons/pcb2blender_importer/site-packages/ python3
Python 3.12.4 (main, Jun 12 2024, 19:06:53) [GCC 13.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import importlib.util
>>> importlib.util.find_spec("skia")
ModuleSpec(name='skia', loader=<_frozen_importlib_external.ExtensionFileLoader object at 0x7f8e5f39baa0>, origin='/home/blackbit/.venv/blender_deps/lib/python3.12/site-packages/skia.cpython-312-x86_64-linux-gnu.so')
```

Does this provide any insights?

>  If you launch Blender normally (without --python-use-system-env) and open a python console in it, what does sys.executable evaluate to?

```
PYTHON INTERACTIVE CONSOLE 3.11.9 (main, Apr 10 2024, 13:16:36) [GCC 13.2.0]

Builtin Modules:       bpy, bpy.data, bpy.ops, bpy.props, bpy.types, bpy.context, bpy.utils, bgl, gpu, blf, mathutils
Convenience Imports:   from mathutils import *; from math import *
Convenience Variables: C = bpy.context, D = bpy.data

>>> import sys
>>> sys.executable
'/home/blackbit/.venv/blender_deps/bin/python'

>>> 
```

```
lrwxrwxrwx 1 blackbit blackbit    7 2024-06-23 22:50 python -> python3
lrwxrwxrwx 1 blackbit blackbit   16 2024-06-23 22:50 python3 -> /usr/bin/python3
lrwxrwxrwx 1 blackbit blackbit    7 2024-06-23 22:50 python3.12 -> python3
```

```
$ /usr/bin/python3 -V
Python 3.12.4
```

30350n, 2024-06-26:
> ```
> >>> import sys
> >>> sys.executable
> '/home/blackbit/.venv/blender_deps/bin/python'
> ```

That looks like your environment variables are still messed up, it shouldn't use that venv by default.

> ```
> $ PYTHONPATH=$HOME/.venv/blender_deps/lib/python3.12/site-packages/:$HOME/.config/blender/4.1/scripts/addons/pcb2blender_importer/site-packages/ python3
> Python 3.12.4 (main, Jun 12 2024, [...]

Not really, as you are using your systems python version. Try evaluating `importlib.util.find_spec("skia")` from within Blender too.

blackbit42, 2024-06-26:
> That looks like your environment variables are still messed up, it shouldn't use that venv by default.

Any ideas?

```
YTHON INTERACTIVE CONSOLE 3.11.9 (main, Apr 10 2024, 13:16:36) [GCC 13.2.0]

Builtin Modules:       bpy, bpy.data, bpy.ops, bpy.props, bpy.types, bpy.context, bpy.utils, bgl, gpu, blf, mathutils
Convenience Imports:   from mathutils import *; from math import *
Convenience Variables: C = bpy.context, D = bpy.data

>>> import importlib.util
>>> importlib.util.find_spec("skia")
>>> 
```

that is from within blender.

30350n, 2024-06-26:
> Any ideas?

Is PYTHONPATH or BLENDER_SYSTEM_PYTHON still set? If so unset those.

>```
> >>> import importlib.util
>>> importlib.util.find_spec("skia")
>>> 
> ```
>
>that is from within blender.

That's the problem. I'm guessing `from skia import SVGDOM, Color4f, Stream, Surface` doesn't here too?

blackbit42, 2024-06-26:
```
raceback (most recent call last):
  File "<blender_console>", line 1, in <module>
ModuleNotFoundError: No module named 'skia'

>>> 
```

This is consistent with the statement the GUI makes.
I am rather puzzled why after gradually installing other dependencies they were detected by the python environment inside blender correctly but skia isn't visible to blender.

blackbit42, 2024-07-18:
Any thoughts, @30350n ?

30350n, 2024-09-21:
Closing this, as it should be resolved by installing the newest [v2.13](https://github.com/30350n/pcb2blender/releases/tag/v2.13-k8.0-b4.2) version which uses the new [Blender Extension Format](https://extensions.blender.org/about/)!

closed 2024-09-21 by 30350n