30350n/pcb2blender #66: Exception when dielectric layer is locked

duddie, 2024-06-26:
The exception happens when dielectric layer is locked:

```
Traceback (most recent call last):

File "/Users/duddie/Documents/KiCad/8.0/3rdparty/plugins/com_github_30350n_pcb2blender/__init__.py", line 24, in Run
export_pcb3d(dialog.file_picker.GetPath(), boarddefs)

File "/Users/duddie/Documents/KiCad/8.0/3rdparty/plugins/com_github_30350n_pcb2blender/export.py", line 108, in export_pcb3d
file.writestr(f"{LAYERS}/{LAYERS_STACKUP}", get_stackup(board).pack())

File "/Users/duddie/Documents/KiCad/8.0/3rdparty/plugins/com_github_30350n_pcb2blender/export.py", line 225, in get_stackup
stackup.thickness_mm = sum(float(match.group(1)) for match in matches)

File "/Users/duddie/Documents/KiCad/8.0/3rdparty/plugins/com_github_30350n_pcb2blender/export.py", line 225, in <genexpr>
stackup.thickness_mm = sum(float(match.group(1)) for match in matches)

ValueError: could not convert string to float: '1.51 locked'

```

[image]

This is caused by enabling dielectric lock in Board Setup:
[image]

By clicking the check box in the **PADLOCK** column.

And this adds in the <project>.kicad_pcb the '**locked**' in the thickness board. 
```
(layer "dielectric 1"
				(type "core")
				(color "#202020FF")
				(thickness 1.51 locked)
				(material "FR4")
				(epsilon_r 4.5)
				(loss_tangent 0.02)
			)

```

Unlocking dielectric makes the plugin works perfect.

30350n, 2024-06-26:
Thanks a lot for the detailed report, this should be fixed in the next release.

duddie, 2024-06-28:
Thanks a lot. Your tool is fantastic! have a nice day!

closed 2024-07-17 by 30350n