30350n/pcb2blender #69: No module named 'io_scene_x3d' in Blender

wjklimek1, 2024-07-19:
Hi,
I installed importer in Blender 4.2 LTS on Linux Mint 21.3. I enabled addon, installed skia-python and pillow with a button in Blender addon UI and restarted Blender. I don't see option to import PCB and when I run Blender from command line it prints following errors:

`Traceback (most recent call last):
  File "/home/wojtek/Programy/blender/4.2/scripts/modules/addon_utils.py", line 401, in enable
    mod = importlib.import_module(module_name)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/wojtek/Programy/blender/4.2/python/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/wojtek/.config/blender/4.2/scripts/addons/pcb2blender_importer/__init__.py", line 26, in <module>
    register, unregister = register_modules_factory(modules)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/wojtek/.config/blender/4.2/scripts/addons/pcb2blender_importer/blender_addon_utils/register.py", line 37, in register_modules_factory
    _modules.append(importlib.import_module(f".{module}", package=caller_package))
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/wojtek/Programy/blender/4.2/python/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/wojtek/.config/blender/4.2/scripts/addons/pcb2blender_importer/importer.py", line 15, in <module>
    from io_scene_x3d import (
ModuleNotFoundError: No module named 'io_scene_x3d'
Saved session recovery to "/tmp/quit.blend"
`
It looks like some dependency is missing. It works in Blender 4.1.1 though. It seems that io_scene_x3d was bundled with Blender 4.1.1 but was not shipped with 4.2. I tried adding https://extensions.blender.org/add-ons/web3d-x3d-vrml2-format/ but installing it in 4.2 did not help. 

I would prefer to use 4.2 LTS release of Blender so I would greatly appreciate any help with this.

30350n, 2024-07-19:
Blender 4.2 is not supported yet.

30350n, 2024-09-21:
Blender 4.2 is supported now in [v2.13](https://github.com/30350n/pcb2blender/releases/tag/v2.13-k8.0-b4.2)!

closed 2024-09-21 by 30350n