30350n/pcb2blender #71: Unable to import .pcb3d file to blender

Georgeeden, 2024-07-27:
I am unable to import the file to blender after installing the add-in.

[image]
[image]
[image]

Any ideas on how to resolve this would be appriciated (Hopefully it is me just doing something wrong, I am still new to blender).

30350n, 2024-07-28:
Blender 4.2 is not supported yet.

closed 2024-07-28 by 30350n

Georgeeden, 2024-07-28:
Thanks, for anyone wanting to install the correct Blender version, it can be found here: https://download.blender.org/release/Blender4.1/blender-4.1.1-windows-x64.msi

30350n, 2024-09-21:
Blender 4.2 is supported now in [v2.13](https://github.com/30350n/pcb2blender/releases/tag/v2.13-k8.0-b4.2)!