30350n/pcb2blender #72: Unable to install addon

homusman, 2024-08-15:
Attempted to install the adon via the usual Blender->Preferences->Addons->Install Zip

Blender says:
"Modules Installed () from 'C:\\Users\\janej\\Downloads\\pcb2blender_exporter_2-12.zip' into 'C:\\Users\\janej\\AppData\\Roaming\\Blender Foundation\\Blender\\4.0\\scripts\\addons' "
But the pcb2blender doesnt actually appear.

Tested in Blender 3.6 4.0 and 4.2.
Blender was restarted multiple times, machine was restarted too.

30350n, 2024-08-15:
pcb2blender currently only supports Blender 4.1.1.

closed 2024-08-15 by 30350n

30350n, 2024-09-21:
Blender 4.2 is supported now in [v2.13](https://github.com/30350n/pcb2blender/releases/tag/v2.13-k8.0-b4.2)!

homusman, 2024-10-06:
Thank you! works like a charm. Would suggest leaving a note in the release notes as i did not find the compatibility info anywhere else.