30350n/pcb2blender #74: Error importing .PCB3D file into Blender.

FraserElectronics, 2024-09-23:
I am using the latest version of KiCad with the Export to Blender plugin. I export a simple PCB to disk and then try and import it into Blender 4.2.1.0 but the following error message is displayed:

[image]

Am I missing a step or what might be wrong?

Andy

30350n, 2024-09-23:
Ah, seems like that happens if you aren't in `Object Mode` when importing. Switching to `Object Mode` before should fix it.

FraserElectronics, 2024-09-23:
Thanks for that 👍
I can now import my PCB however, I notice there are no tracks or Silkscreen - is that to be expected?

Andy

30350n, 2024-09-23:
Try to switch to `Rendered` view mode.

FraserElectronics, 2024-09-23:
Okay, that is really cool.
Very nice plugin 👍

Andy

closed 2024-09-23 by FraserElectronics

30350n, 2024-09-23:
Thanks!

The error when importing in edit mode will be fixed in the next version ^^