30350n/pcb2blender #75: Doesn' work with Kicad 8.99

positroniumx, 2024-10-01:
I installed the add-on via Kicad's plugin manager. It detects the board but errors when exporting the file. I tried on both Windows and Linux versions of Kicad, same error.

[image]

30350n, 2024-10-03:
Thanks for the note, but I won't look into this further until KiCad 9.0 is stable.

positroniumx, 2024-10-03:
Wouldn't have mentioned it but since I saved my project files in 8.99 format I can't open them with the previous version. Maybe a heads up for anyone starting a new project with the hopes of using this add-on, use an older version of Kicad. Or maybe add a version compatibility notice.

30350n, 2024-10-03:
I've set the `kicad_version_max` to 8.0 now in the `metadata.json`, so newer versions shouldn't be displayed in the Package Manager anymore (althought for this to really make sense I'd have to also do this for all the other old versions).

Also, x.99 versions are development versions not meant for production use afaik.

eriklundh, 2024-10-03:
We are already at Kicad 8.0.5 
Your setting of kicad_version=8.0 might be why 2.14 never showed up  in my 8.0.5 Kicad Package Manager. I had to download the 2.14 exporter as a zip file and install the plugin from disk.
Dev versions seems to be numbered 8.99
8.98 might be a better value for kicad_max_version.

30350n, 2024-10-03:
> Your setting of kicad_version=8.0 might be why 2.14 never showed up in my 8.0.5 Kicad Package Manager.

I literally released v2.14 3 hours ago, even if I would have submitted a PR to the PCM repo, it wouldn't have been merged yet, so not sure what you are on about. (Also I won't add v2.14 to the PCM repo, because it's functionally identical to v2.13 and yes that's also the reason why some other versions are missing.)

> Your setting of kicad_version=8.0 might be why 2.14 never showed up in my 8.0.5 Kicad Package Manager.
> 8.98 might be a better value for kicad_max_version.

No and no ([*RTFM*](https://dev-docs.kicad.org/en/addons/index.html#_fields)).

1. `kicad_version` refers to the minimum KiCad version supported, specifying a `PATCH` version is not required. 
2. `kicad_version_max` refers to the last version that's compatible (once again specifying a `PATCH` version is not required).

So setting both to `"8.0"` should be perfectly fine.

positroniumx, 2024-10-03:
> I've set the `kicad_version_max` to 8.0 now in the `metadata.json`, so newer versions shouldn't be displayed in the Package Manager anymore (_althought for this to really make sense I'd also have to do this for all the other old versions I guess_).
> 
> Also, x.99 versions are development versions not meant for production use afaik.

Yeah these things come with the territory of using bleeding edge, I used kicad-git package. Just that I lost some time trying to make it work thinking it's something on my end, since I got the add-on from the plugin manager. Didn't expect version issues.

30350n, 2024-10-03:
> Didn't expect version issues.

Yeah this in particular just seems like they renamed/removed some arguments of `ExportVRML`. But I hope you understand that I really can't be bothered to be up to date with every little change like that ...

Will keep this open to remind me to fix this at some point:
> set the `kicad_version_max` to 8.0 [...] *for this to really make sense I'd also have to do this for all the other old versions I guess*

positroniumx, 2024-10-03:
Yeah I understand no problem, just mentioned it as a heads up for others.

eriklundh, 2024-10-03:
My mistake, I was thinking of kicad_version_max.
Good to know that patch versions are insignificant.

30350n, 2024-11-06:
Closing this, [v2.15](https://github.com/30350n/pcb2blender/releases/tag/v2.15-k8.0-b4.2) should clarify that nightly builds are not supported.

closed 2024-11-06 by 30350n