30350n/pcb2blender #76: Python error on linux

fami-fish, 2024-10-06:
Extension wont activate:
[image]

Pip is installed:
[image]

Any help as to how i can fix this? I installed pillow on my system too.. and blender's own pip environment seems to have PIL installed.

30350n, 2024-10-06:
Not sure what's the issue here tbh, only thing that seems wrong is your python version. It should be 3.11 but is 3.12 for some reason (incorrect packaging most likely).

Looking at that installation path, it could in theory also be a dependency collision with some other extension you installed.

fami-fish, 2024-10-08:
> Not sure what's the issue here tbh, only thing that seems wrong is your python version. It should be 3.11 but is 3.12 for some reason (incorrect packaging most likely).
> 
> Looking at that installation path, it could in theory also be a dependency collision with some other extension you installed.

Is there a breaking change in 3.12 from 3.11? It should still run i assume the version bump does not include breaking changes.

Also, its not a dependency collision as i tried removing my plugins (of which there was only one anyways, which im pretty sure did not utilize pip). Running ls on the `.local/lib/python3.12/site-packages/PIL` directory shows its completely empty, and the directory and the `__init__.py` file does not exist. Any guidance? im going to do a system upgrade now in hopes that it fixes things but maybe a way to manually install pip and/or symlink it to the directory would be good. I still have no clue how blender is seeing _some_ version of pil, even though its not present in .local anywhere

30350n, 2024-10-08:
> Is there a breaking change in 3.12 from 3.11? It should still run i assume the version bump does not include breaking changes.

Generally no, but Blender not using the correct python version generally hints that whoever packaged Blender for whatever distro you are using messed up. It could for example be that they decided to reuse your system Python for Blender, in which case, It could be that and outdated version of `pillow` (or worse, `PIL`) is installed there.

It could also be that the `pillow` wheel included with the extension (`pillow-10.4.0-cp311-cp311-manylinux_2_28_x86_64.whl` from [here](https://pypi.org/project/pillow/10.4.0/#files)) somehow fails to install because of the version mismatch.

babolait, 2024-10-21:
I have the same problem under blender 4.2.1 :

[image]

Sorry, newbie here, but, you need to install pip in order to run the module pcb importer ?

When I run the commands of @fami-fish (PIL / print(PIL.__version__) it seem "PIL" is not installed : 
 
[image]

On the first try of enabling pcb importer the problem was "no module named skia", after reboot of blender, it is now the PIL error.

I am on fedora 40.

30350n, 2024-10-21:
Like I said previously, this is most likely a packaging issue, not an issue with this extension or Blender. In the case of Fedora 40, it almost definitely is:

[image]

So either install Blender by directly downloading a build from https://www.blender.org/download/ or possibly by using something like [flatpak](https://flatpak.org/).

The other option would be to bug your responsible package maintainer to fix their packaging to produce non broken Blender builds.

mothdotmonster, 2024-10-25:
I can confirm this issue also occurs for me on Arch Linux.

babolait, 2024-10-25:
> Like I said previously, this is most likely a packaging issue, not an issue with this extension or Blender. In the case of Fedora 40, it almost definitely is: [...]

I tried with the version downloaded directly from blender website. I am new to fedora / linux, so not sure if I have done it correctly. I tried installing your module and get the same error. 
After that I installed it using flatpak. It is the 4.11. So when installed, I installed the dependencies (doing .venv and other console lines as asked) of blender (I assume they are needed for your module ?), using their tutorial.

I went for the install of your module, it could not load it cause of the init.py being at the top level, I created a folder and re-zipped it. Blender took the module, and said "module at XX installed at XXX" (see attachment), but when I search for it in the list, impossible to find it, do you have an idea ? I refreshed it, reboot the computer, re-do the installation but impossible to get it in the list and enable it. 

[image]

I am sorry if those question are trivial, I have very hard time finding info on the install ;)
I can post that message elsewhere if you want.
Thank you

30350n, 2024-10-27:
> I tried with the version downloaded directly from blender website. I am new to fedora / linux, so not sure if I have done it correctly. I tried installing your module and get the same error.

Can you share a screenshot of that again, aswell as a screenshot of your python console, as before?

> After that I installed it using flatpak. It is the 4.11.

I guess you mean 4.1.1 here? The current version of this plugin only supports 4.2+ so that's not going to work.

> So when installed, I installed the dependencies (doing .venv and other console lines as asked) of blender (I assume they are needed for your module ?), using their tutorial.

Not sure what that's supposed to mean exactly, but none of that should be required.

Installation should be as simple as downloading Blender and installing the `PCB Importer` extension via the builtin extensions UI.

babolait, 2024-10-31:
Thank you for your answer. 

> > I tried with the version downloaded directly from blender website. I am new to fedora / linux, so not sure if I have done it correctly. I tried installing your module and get the same error.
> 
> Can you share a screenshot of that again, aswell as a screenshot of your python console, as before?

In fact my problem is that I still don't get very well how to install stuff in fedora / linux. 
When I tried with direct source on blender, I had a folder with a .exe, and when launching the app it work well but I was searching for a way to "install" it (have the app show up in the gnome), so I stopped here.

As for now, I launch the .exe and it work. After that your module that I had already installed showed up and I activated it and then imported the kicad generated file and it work like a charm, on fedora / blender 4.2.3 LTS.

On my render, do you know why there is black lines on my pcb ?

[image]

30350n, 2024-10-31:
> On my render, do you know why there is black lines on my pcb ?

Because you are only previewing in viewport render mode. (Those lines represent relationships between objects).

asdf23, 2024-12-05:
Okay same issue Fedora 40 user. I tried installing the missing packages in python 3.12 this didn't work. Flatpak did work.

KaiMartin, 2025-04-19:
I experience a similar issue in Debian trixie/testing: The importer can be installed via the blender GUI. But on activation I just get an error message:
   RuntimeError: Extension bl_ext.blender_org.pcb3d_importer is incompatible (This Python version (3.13) isn't compatible with (3.11))
The debian package depends on libpython 3.13 (>=3.13.0). So just like arch from the bug report above, blender as distributed by Debian depends on the systems version of python3. Given the multitude of python dependencies in blender, there is little chance that Debian would change their packaging policy. They'd essentially have to maintain an additional set of python libs just for blender. 
Is there any way to bump the python version of the importer add-on?

KaiMartin, 2025-04-20:
> Is there any way to bump the python version of the importer add-on?

I tried to clone the project and bump the required version of python3 myself. Unfortunately, getting the plugin to load  in blender turned out to be beyond my skill level.

So I reluctantly uninstalled the Debian package  and downloaded a blender binary directly from blender.org . As hinted above, this binary comes with python 3.11 . Consequently, it successfully loads the plugin. My PCBs rendered beautifully. Good job on the choice of colors and materials!

30350n, 2025-04-20:
Yeah, a lot of popular linux distros are sadly incapable of correctly packaging Blender, shipping wrong python versions.

Glad you got it figured out and thank you for the kind words!