30350n/pcb2blender #78: 3D models with relative paths not supported

hraftery, 2024-11-04:
My KiCad footprint libraries use [this feature](https://gitlab.com/kicad/code/kicad/-/issues/2073), which was added a couple of years ago. To wit, the path to the 3D model for a footprint in my library is relative to the location of the footprint.

It seems this triggers:

> Failed to retrieve file times for '' (error 2: No such file or directory)"

when running the plugin. The message is repeated 8 times, which happens to be the number of 3D models in my project.

The output file is still generated, and it even has 3d models, but... they're all the same!

Manually updating each footprint to an absolute path gives the expected results.

PS. Awesome plugin, much appreciated.

30350n, 2024-11-05:
Hi, thanks for reporting this, I think this might not be an issue with this addon though.

All the exporter does is basically call KiCad's VRML exporter with `Copy 3D model files to 3D model path` enabled, serialize a bunch of other metadata and pack everything into a zip file.

I'm fairly confident that you'll get the same errors if you just use the VRML exporter from the PCB Editor UI. If so, it'd be best if you'd report this at https://gitlab.com/kicad/code/kicad/-/issues.

hraftery, 2024-11-05:
Right you are. And I found the issue [already exists](https://gitlab.com/kicad/code/kicad/-/issues/14505).

Cheers.

set-soft, 2024-11-05:
> Right you are. And I found the issue [already exists](https://gitlab.com/kicad/code/kicad/-/issues/14505).
> 
> Cheers.

Yes, and I'll suggest to @30350n to upvote it, I reported it for KiCad 7 and Seth tagged it as "low-priority" (most probably because he didn't understand how bad the bag is)

GreenCappuccino, 2025-04-11:
Hello, I submitted [a patch](https://gitlab.com/kicad/code/kicad/-/merge_requests/2211) that just got merged that should resolve the export error. May want to see if this works in the next release the commit makes its way into.

30350n, 2025-04-12:
Thanks a lot for looking into this! Doesn't seem like it got backported to 9.0 though, so we might have to wait for a while.

hfieger, 2025-04-14:
While exporting from KiCad 9 i receive this errors:

[image]

And i get this result in Blender 4.4:

[image]

It looks like, the last valid part from the list will be duplicated instead of the right one.

Does this belong  to this thread?

hraftery, 2025-04-14:
Looks exactly like the symptoms in my original report, so I'd say yes.

hfieger, 2025-04-14:
Thanks for your answer. I oversaw your first comment.

I have all my local stp-files in one directy and copied them temporarly to the KiCad project directy.
Now the export function works. Maybe this solution helps other persons who need a quick solution.

Best regards

closed 2025-05-24 by 30350n